riscv-collab/riscv-gnu-toolchain #431: recipe for target 'configure-gcc' faild

minamagdy00, 2019-05-01:
Hello all,
I've the following issue during my first build for the riscv toolchain:

configure: error: uint64_t and int64_t not found
Makefile:4203: receipe for target 'configure-gcc' failed.

My current used directory is the cloned riscv-gnu-toolchain and using the ./configure --prefix=/any_install_path then make.

So how to solve that error ?.

minamagdy00, 2019-05-01:
* my gcc version is 7.4.0 and ubuntu 18.4 release

jim-wilson, 2019-05-02:
A confusing error, but with a bit of web searching, I figured out that I can reproduce the error if I don't have a C++ compiler installed.  So the solution is to install a C++ compiler.  E.g. "apt install g++" or equivalent.

The normal way to debug a configure problem like this is to look for the relevant error in the config.log file.  Depending on how far configure got, there may be more than one config.log file, so make sure you are looking at the one in the right directory, which in this case would be the one in the gcc subdir.  Then look for the "uint64_t or int64_t not found" error and work your way backwards.  I expect you will see errors for a missing C++ compiler.

There is already an FSF GCC bug report for this, reported four and a half years ago.
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=63509

closed 2019-05-16 by jim-wilson