riscv-collab/riscv-gnu-toolchain #434: Modifying save-restore.S

JoshSabet, 2019-05-14:
I am trying to edit the compiler to use the instructions I have created. I am interested in changing the way registers are saved to the stack and are restored. To do this, should I modify the file riscv-gnu-toolchain/riscv-gcc/libgcc/config/riscv/save-restore.S ? 

Also if I understand correctly:
-  __riscv_save_12 indicates register s11 should be saved, __riscv_restore_11 indicates register s10 should be loaded
- .cfi_return_column 5 stores the return address in t0
- .cfi_def_cfa_offset 112 allocates 112 bytes on the stack pointer
- .cfi_offset 27, -104 stores register s11 -104 bytes from the stack pointer

I am not sure if I understand what .cfi_restore does

aswaterman, 2019-05-14:
save-restore.S is only used when the -msave-restore option is passed to the compiler.  You're probably more interested in looking at riscv_expand_prologue and riscv_expand_epilogue, in riscv-gnu-toolchain/riscv-gcc/gcc/config/riscv/riscv.c.

jim-wilson, 2019-05-16:
cfi is call frame information used for stack unwinding, which is turn is used for C++/etc exception handling.  The cfi pseudo ops describe the stack frame that a function creates, the size, the location of saved registers, etc.  cfi_restore says a register is back in its original state, i.e. not on the stack.

  .cfi_def_cfa_offset 112
says the frame has size 112 bytes.
  .cfi_offset 27, -104
says the original value of register 27 (s11) is located -104 bytes from the top of the frame, or sp+8.
  ld s11, 8(sp)
this loads the original value of s11 back into s11 from its stack slot.
  .cfi_restore 27
This says that the original value of s11 is now back in s11.

The routines store 2 registers at a time, and stack.  This helps to reduce the code size, and helps keep the stack aligned.  So __riscv_save_12 indicates that s0 through s11 will be saved on the stack.  __riscv_restore_11 indicates that s10 through s0 will be reloaded from the stack.  Also, __riscv_restore_10 does the same thing as __riscv_restore_11.  This means in 1/2 the cases that we save/restore one extra register, but that is harmless.

As Andrew mentioned, none of this is relevant to the normal support for saving/restoring registers which is in riscv_expand_prologue and riscv_expand_epilogue.

JoshSabet, 2019-05-22:
I have been looking at the functions that riscv_expand_prologue calls and I have some confusion. Which functions are called from riscv_expand_prologue() to write the instructions that store the registers to the stack. I am guessing it starts at 

riscv_for_each_saved_reg() -> riscv_save_reg() -> riscv_emit_move() -> emit_move_insn() -> emit_move_insn_1 -> emit_insn() -> add_insn()

If I am understanding the code correctly does  this riscv_expand_epilogue() discover which registers must be saved, and then stores them onto a doubly linked list?

jim-wilson, 2019-05-22:
riscv_compute_frame_info figures out what registers need to be saved, the size of the frame required, and what offsets stuff will be saved at in the frame.  This function may be called multiple times, because this info can change during optimization.  The info isn't fixed until the last time it is called.

riscv_expand_prologue emits the instructions that build the frame, using the info computed earlier by riscv_compute_frame_info.

There is no doubly linked list here, other than the RTL instruction stream, which is doubly linked so we can move forward and backward through the instruction stream.

JoshSabet, 2019-06-11:
From what I understand, the compiler generates a list of registers to be stored and then generates the instruction code for all of these store words and adds the instructions to the instruction stream. 

If I am correct, the instructions that are to be stored are indicated by cfun->machine->frame->mask

I am having difficulty finding where exactly the instruction code is generated however, as I would need the compiler to use my instructions and I would like to see how instructions are added to the compiler.

jim-wilson, 2019-06-11:
See riscv_for_each_saved_reg.  This calls riscv_save_restore_reg which generates the RTL using the function passed down which will be either riscv_save_reg or riscv_restore_reg depending on whether we were called from riscv_expand_prologue or riscv_expand_epilogue.

JoshSabet, 2019-06-19:
So riscv_save_reg calls riscv_emit_move which calls emit_move_insn_1

This function has the statement "code = optab_handler (mov_optab, mode);". Is this where the code for the instruction is determined?

jim-wilson, 2019-06-19:
optab_handler just indexes into mov_optab using the mode.   The target backend registers every mov* pattern into the mov_optab handler.  If we need a move of a particular mode, and the backend supports it, then we emit that via GEN_FCN.  If the backend doesn't support it, then we can try alternate strategies.  For instance, if we need a DImode move and the target doesn't support that, but it does support SImode moves, then perhaps we can emit 2 SImode moves to get the same result.

If you are trying to emit a different kind of instruction, then this is probably the wrong place, as we have already decided what kind of instruction we want here, and the only question is whether the target supports it and we can emit it directly, or whether we have to synthesize it from other instructions that the target does support.  To get a different kind of instruction, you need to change something higher up before we get here.

JoshSabet, 2019-06-19:
In that case, at which point do we decide which instruction to emit?

jim-wilson, 2019-06-20:
In riscv_expand_prologue, we obviously want to emit word-sized store instructions.  We iterate through the list of registers that need to be saved in riscv_for_each_saved_reg, and then emit the store instruction via riscv_save_reg for each one.  riscv_save_reg was passed in from above as fn.

In riscv_expand_epilogue, we obviously want to emit word-sized load instructions.  We iterate through the list of registers that need to be restored in riscv_for_each_saved_reg, and then emit the load instruction via riscv_restore_reg for each one.  riscv_restore_reg was passed in from above as fn.

You would probably make faster progress if you explained what you are trying to do, rather than ask random disconnected questions.

If you have an instruction that can save/restore two regs at a time for instance, then you want to look at riscv_for_each_saved_reg, and try to handle two adjacent registers at a time instead of one at a time.  Or maybe replace the call to riscv_for_each_saved_reg with a call to your own function.

JoshSabet, 2019-06-20:
The instruction I wish to implement would store all specified registers in R[rs2] into M[R[rs1] + imm] (my instruction is of S Type). They are stored as a bit vector where bit 0 corresponds to x0, etc.

So I would like to use the value frame->mask as the value in rs2

JoshSabet, 2019-07-03:
Since the issue has changed dramatically, I have decided to close it and open a new issue.

closed 2019-07-03 by JoshSabet