riscv-collab/riscv-gnu-toolchain #436: Canadian Toolchain Build

fkladdu, 2019-05-15:
Hi jim

can you please tell How Canadian Tool is Build, if there is any procedure is there Tell me.

Thanks 
khan.

fkladdu, 2019-05-15:
canadian Toolchain for Riscv

kito-cheng, 2019-05-15:
Are you want to build a mingw toolchain on linux?

fkladdu, 2019-05-15:
yess...

kito-cheng, 2019-05-15:
Just reply again to make sure you will got mail again, I've edited last comment to add how to build canadian toolchain.

fkladdu, 2019-05-15:
How can i get x86_64-w64-mingw32-gcc this Tool?

fkladdu, 2019-05-15:
Building a canadian Toolchain 

cloned riscv-gnu sources

$ git clone --recursive https://github.com/riscv/riscv-gnu-toolchain

Here i have gone through Few trials with riscv sources.

Try1:
----------
$ sudo apt-get install mingw-w64
$ ./configure --with-host=x86_64-w64-mingw32 --prefix=<path> 
$ make linux

Error:
------------
checking for the correct version of gmp.h... no
configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
ftp://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.
Makefile:276: recipe for target 'stamps/build-gcc-linux-stage1' failed
make: *** [stamps/build-gcc-linux-stage1] Error 1

Try to solve Error:
------------------------
Downloaded ftp://gcc.gnu.org/pub/gcc/infrastructure/.  sources
$ ./configure --with-host=x86_64-w64-mingw32 --with-gmp=<path_gmp-4.3.2>  --prefix=<path> 

Here its Not recognising gmp libraries.
----------------------------------------------------------------------------------------------------------------------
Try2:
----------
$ sudo apt-get install mingw-w64
$ ./configure --with-host=x86_64-w64-mingw32 --prefix=<path> 
$ make linux

Error:
----------
configure: error: expat is missing or unusable
Makefile:9104: recipe for target 'configure-gdb' failed
make[2]: *** [configure-gdb] Error 1

Try to solve Error:
-------------------------
$ sudo apt-get install expat
$ ./configure --with-host=x86_64-w64-mingw32 --prefix=<path> 
$ make linux

Still getting same Error.

kindly, help me to resolve this Error.

jim-wilson, 2019-05-15:
This was asked both on sw-dev and here.  I pointed at crosstool-ng in my sw-dev reply.

For try one, you need to build windows versions of the gmp/mpc/mpfr libraries to use for building the windows compiler.

For try two, you need  to build a windows version of the expat libraries to use for building the windows gdb.  if you don't care about the windows hosted gdb, then you can turn off the expat support.  There are hard coded --with-expat=yes options in the toplevel makefile you can remove.

In general, I wouldn't recommend using riscv-gnu-toolchain for a canadian cross build.  But if you want to try it, you will need to spend time analyzing the build failures and figuring out how to fix them.  There is a config.log file in every directory that has a configure script, ,that will have details of what went wrong when configuring.  If you hit a build error, then you should save a build log and look at the build log to see what went wrong.

jim-wilson, 2019-05-16:
Answered on sw-dev.

closed 2019-05-16 by jim-wilson