riscv-collab/riscv-gnu-toolchain #442: Building Riscv toolchain on Macos make linux error

snathicktechno, 2019-05-17:
I am trying to build Riscv toolchain on macos 
my macos version 10.14.4 (macos Mojave)

./configure= /toolchain/intsallation/path
make linux 

sed > /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/shlib.ldsT \

      -e '/^=========/,/^=========/!d;/^=========/d' \

      -e 's/^.*\.gnu\.hash[ ]*:.*$/  .note.ABI-tag : { *(.note.ABI-tag) } &/' -e '/^[ ]*\.hash[
]*:.*$/{h;d;}' -e '/DATA_SEGMENT_ALIGN/{H;g}'  \

      -e 's/^.*\*(\.dynbss).*$/& \

PROVIDE(__start___libc_freeres_ptrs = .); \

*(__libc_freeres_ptrs) \

PROVIDE(__stop___libc_freeres_ptrs = .);/'\

      -e 's@^.*\*(\.jcr).*$@& \

PROVIDE(__start___libc_subfreeres = .);\

__libc_subfreeres : { *(__libc_subfreeres) }\

PROVIDE(__stop___libc_subfreeres = .);\

PROVIDE(__start___libc_atexit = .);\

__libc_atexit : { *(__libc_atexit) }\

PROVIDE(__stop___libc_atexit = .);\

PROVIDE(__start___libc_thread_subfreeres = .);\

__libc_thread_subfreeres : { *(__libc_thread_subfreeres) }\

PROVIDE(__stop___libc_thread_subfreeres = .);\

PROVIDE(__start___libc_IO_vtables = .);\

__libc_IO_vtables : { *(__libc_IO_vtables) }\

PROVIDE(__stop___libc_IO_vtables = .);\

/DISCARD/ : { *(.gnu.glibc-stub.*) }@'

sed: 1: "/DATA_SEGMENT_ALIGN/{H;g}

": extra characters at the end of g command

make[3]: *** [/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/shlib.lds] Error 1

make[2]: *** [elf/subdir_lib] Error 2

make[1]: *** [all] Error 2

make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2

can anyone help me to resolve this error

Thanks
snathick

aswaterman, 2019-05-17:
Looks like you need to use a different sed (GNU sed).

I also seem to recall that glibc can only build with a case-sensitive file
system, though it is possible they’ve fixed that limitation. You might have
an easier time installing a Linux VM and building there instead.

ilg-ul, 2019-05-17:
Building GCC with the Apple development environment has always been a challenge. I don't know on Mojave, but on previous versions it was not practical.

The common workaround is to install the required GNU tools and libraries via HomeBrew.

snathicktechno, 2019-05-20:
Thank for your suggestions aswaterman and ilg-ul.
As per aswaterman i installed GNU-sed

Again trying to build Riscv toolchain on macos
my macos version 10.14.4 (macos Mojave)

./configure --prefix= /toolchain/intsallation/path
make linux

i got below error

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/csu/crt1.o: in function `.L0 ':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/csu/../sysdeps/riscv/start.S:50: undefined reference to `__libc_csu_init'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/csu/../sysdeps/riscv/start.S:52: undefined reference to `__libc_csu_fini'

collect2: error: ld returned 1 exit status

make[3]: *** [/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/iconv/iconvconfig] Error 1

make[2]: *** [iconv/others] Error 2

make[1]: *** [all] Error 2

make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2

can anyone help me to resolve this error

Thanks
snathick

ilg-ul, 2019-05-20:
As the name implies, `riscv64-unknown-linux-gnu` is the native Linux toolchain, why do you think it should compile natively on macOS?

snathicktechno, 2019-05-20:
Thanks for the reply ilg-ul

my macos version 10.14.4 (macos Mojave)

I am trying to build a toolchain with the following:

build:macOS
host:macOS
target:RISCV linux

Again trying to build Riscv toolchain on macos

./configure --with-host=x86_64-apple-darwin18.6.0 --prefix=/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install

make linux

iam getting following errors 

Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc_pic.os: in function `printf_positional':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/stdio-common/vfprintf.c:2023: undefined reference to `__readonly_area'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc_pic.os: in function `_IO_vfprintf_internal':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/stdio-common/vfprintf.c:1325: undefined reference to `__readonly_area'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc_pic.os: in function `printf_positional':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/stdio-common/vfprintf.c:1878: undefined reference to `__readonly_area'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/stdio-common/vfprintf.c:2022: undefined reference to `__readonly_area'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc_pic.os: in function `gaiconf_init':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/posix/../sysdeps/posix/getaddrinfo.c:2109: undefined reference to `__GI___fortify_fail'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/posix/../sysdeps/posix/getaddrinfo.c:2158: undefined reference to `__GI___fortify_fail'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/posix/../sysdeps/posix/getaddrinfo.c:1978: undefined reference to `__GI___fortify_fail'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/posix/../sysdeps/posix/getaddrinfo.c:1875: undefined reference to `__GI___fortify_fail'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc_pic.os: in function `__euidaccess':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/io/../sysdeps/posix/euidaccess.c:121: undefined reference to `__GI___vfprintf_chk'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc_pic.os: in function `wctype_table_lookup':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/wctype/wchar-lookup.h:72: undefined reference to `__stack_chk_fail'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc_pic.os: in function `__msgwrite':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/sunrpc/clnt_unix.c:514: undefined reference to `__GI___chk_fail'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc_pic.os: in function `__GI_clnt_spcreateerror':

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/riscv-glibc/sunrpc/clnt_perr.c:279: undefined reference to `__GI___chk_fail'

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc.so: hidden symbol `__readonly_area' isn't defined

/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/8.3.0/../../../../riscv64-unknown-linux-gnu/bin/ld: final link failed: bad value

collect2: error: ld returned 1 exit status

make[3]: *** [/Users/snathick/riscv-toolchain/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/libc.so] Error 1

make[2]: *** [elf/subdir_lib] Error 2

make[1]: *** [all] Error 2

make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2

can anyone help me to resolve this error

Thanks
snathick

ilg-ul, 2019-05-20:
So your goal is to build a cross toolchain that will run on macOS and generate RISC-V Linux binaries?

snathicktechno, 2019-05-21:
Thanks for the response ilg-ul 
yes my goal is to build a cross toolchain that will run on macOS and generate RISC-V Linux binaries.
could you suggest me the steps involved. 

Thank you
snathick

jim-wilson, 2019-05-21:
riscv-gnu-toolchain is rather limited in what it supports.  It can build riscv cross compilers on linux.  It probably can't do anything else.

Normally I would recommend I would recommend crosstool-ng instead, but unfortunately it does not officially support macos.

SiFive has a git repo that we use for our macos builds.  You could try that.  This is github.com/sifive/freedom-tools.  This is known to work on macos, but I'm not the one who does the macos builds so I don't know the details.  I see that it does do a few special things for macos
LIBTOOLIZE ?= glibtoolize
TAR ?= gtar
SED ?= gsed
AWK ?= gawk
This presumably requires HomeBrew.

ilg-ul, 2019-05-21:
> This presumably requires HomeBrew.

I confirm.

try to reconstruct the GNU environment on macOS using HomeBrew. 

as Jim mentioned, in HomeBrew the tools that have identic names with native macOS tools are prefixed with 'g', so you need to tell the configure scripts the new names.

you need to install some libraries and some tools. I do not have a full list, I gave up using HomeBrew since I had lots of problems.

in case it helps, you may take a look at my (now deprecated) script to install HomeBrew, at the end you'll see the packages I used for my build environment.

https://github.com/xpack/xpack-build-box/blob/master/macos/install-homebrew-xbb.sh

another issue to consider is the compiler. you may first try with the Apple clang, but be prepared to see lots of warnings. if you cannot make the build pass, try to install gcc in HomeBrew.

this might work, but I encountered cases where the generated binaries failed to run properly (for example when you compile GDB with GCC on some older macOS versions).

as I said before, for my needs HomeBrew proved inapropriate, I could not create a stable build environment to be used both on a recent macOS and an older one (required to make the generated binaries compatible with older systems too).

the main issue is a design choice, similarly to Arch, HomeBrew is a rolling distribution, continuously advancing all packages at their own pace. totally unusable for a stable build environment.

after lots of experiments, I had no other solution than creating the full build environment (called the xPack Build Box - XBB), and this is what the https://github.com/xpack/xpack-build-box/blob/master/macos/build-xbb.sh script does. 

I don't say that you should use it, if you only want to play with the compiler, you probably can live with HomeBrew, with the disadvantage that from time to time updates may blow your builds. in my case, for production builds, this is not acceptable.

good luck!

snathicktechno, 2019-05-22:
Thanks ilg-ul i will go through it.

lazyparser, 2021-03-04:
> Thanks ilg-ul i will go through it.

Hi @snathicktechno 

Have you gone through it? Is this issue still relevant?

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year and the issues relating to building the toolchain on macOS have been discussed and, it seems, adequately addressed. No RISC-V GCC toolchain issue identified.

closed 2022-09-27 by TommyMurphyTM1234