riscv-collab/riscv-gnu-toolchain #443: undefined reference to `fma' and undefined reference to `__sync_fetch_and_add_4'

Lustrew, 2019-05-19:
```shell
lustre@W:~/new/riscv-sodor/riscv-tests/benchmarks$ make
riscv32-unknown-elf-gcc -I./../env -I./common -I./median -I./qsort -I./rsort -I./towers -I./vvadd -I./multiply -I./dhrystone -I./spmv -I./pmp -I./mm -I./mt-vvadd -I./mt-matmul -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -static -nostartfiles -lgcc -lm -T ./common/test.ld -o mm.riscv ./mm/mm.c ./mm/mm_main.c ./common/syscalls.c ./common/crt.S
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccGw5BKA.o: in function `.L5':
mm.c:(.text+0x15c): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: mm.c:(.text+0x18c): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: mm.c:(.text+0x1bc): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: mm.c:(.text+0x1f4): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccGw5BKA.o: in function `.L7':
mm.c:(.text+0x288): undefined reference to `fma'
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccGw5BKA.o:mm.c:(.text+0xcf0): more undefined references to `fma' follow
/home/lustre/riscv_bin/lib/gcc/riscv32-unknown-elf/8.3.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccSMtsod.o: in function `.L0 ':
mm_main.c:(.text+0x30): undefined reference to `__sync_fetch_and_add_4'
collect2: error: ld returned 1 exit status
Makefile:56: recipe for target 'mm.riscv' failed
make: *** [mm.riscv] Error 1
```
I have change https://github.com/riscv/riscv-tests/blob/master/benchmarks/Makefile#L42 to 
```makefile
RISCV_LINK_OPTS ?= -static -nostartfiles -lgcc -lm -T $(src_dir)/common/test.ld  
```
But it doesn't work. 
Could someone help me!!!

jim-wilson, 2019-05-19:
fma is a math library function. Add -lm to your linker options. This was in RISCV_LINK_OPTS before you modified it.

__sync_fetch_and_add_4 is a 4-byte synchronization primitive. It will be emitted inline if you compile for a target with the A extension. If you compile for a target without the A extension, then it isn't possible to implement it. The problem here is that you can't run a multithreaded testcase on a target without the A extension.

Lustrew, 2019-05-24:
> fma is a math library function. Add -lm to your linker options. This was in RISCV_LINK_OPTS before you modified it.
>
> __sync_fetch_and_add_4 is a 4-byte synchronization primitive. It will be emitted [...]

I am sorry your solution didn't work and I want to know what is A extension?

ilia-diachkov, 2019-05-24:
It's “A” Standard Extension for Atomic Instructions. You can find its description in RISC-V Instruction Set Manual.

closed 2019-10-21 by Lucas-Wye