riscv-collab/riscv-gnu-toolchain #444: The result of FMIN/FMAX instructions when the input operands are +0/-0

hyf6661669, 2019-05-19:
Let's think about 2 special situations:
1. operand1 = -0 = 32'h80000000, operand2 = +0 = 32'h00000000;
2. operand1 = +0 = 32'h00000000, operand2 = -0 = 32'h80000000;

In these situations, what it the right result?

In MATLAB, we have:
```
>> min(single(+0), single(-0))

ans =

  single

   00000000

>> min(single(-0), single(+0))

ans =

  single

   80000000
```

aswaterman, 2019-05-19:
In RISC-V, the FMIN and FMAX instructions treat -0 as smaller than +0, so FMIN(-0, +0) = FMIN(+0, -0) = -0.

This corresponds to the IEEE 754 function minimumNumber. The Matlab behavior corresponds to the older IEEE 754 function minNum, which has weaker semantics than minimumNumber.

jim-wilson, 2019-05-19:
The wikipedia page for IEEE 754-2008 mentions the min and max behavior here.
    https://en.wikipedia.org/wiki/IEEE_754-2008_revision#min_and_max

The new stuff Andrew referred to is in the draft for the next version of the IEEE FP standard, but hasn't been published yet.  Looks like it has been formally submitted and may be an official standard within a few months.
    http://754r.ucbtest.org/

aswaterman, 2019-05-19:
Thanks for the context, Jim. I had forgotten those new routines aren’t part of a ratified standard yet.

hyf6661669, 2019-05-19:
@aswaterman 
@jim-wilson 
Thanks for your answers.

closed 2019-05-19 by hyf6661669