riscv-collab/riscv-gnu-toolchain #446: found error when use make command after "./configure --prefix=/opt/riscv"

mayank2016, 2019-05-21:
When I build the Linux cross-compiler, I run two commands,
./configure --prefix=/opt/riscv
make 

i got error

--------------------------------------
----------------------------------
TARGET_CPU_DEFAULT="" \
HEADERS="auto-host.h ansidecl.h" DEFINES="" \
/bin/sh ../.././riscv-gcc/gcc/mkconfig.sh bconfig.h
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings -fno-common  -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/build -I../.././riscv-gcc/gcc/../include  -I../.././riscv-gcc/gcc/../libcpp/include  \
	-o build/genmodes.o ../.././riscv-gcc/gcc/genmodes.c
as: line 106: exec: -I: invalid option
exec: usage: exec [-cl] [-a name] [command [arguments ...]] [redirection ...]
make[2]: *** [build/genmodes.o] Error 1
make[2]: Leaving directory `/localdata/mayank/new_git/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc'
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory `/localdata/mayank/new_git/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [stamps/build-gcc-linux-stage1] Error 2
[mayank@tejas riscv-gnu-toolchain]$ 

please give solution

thanks & regards
Mayank

jim-wilson, 2019-05-21:
I think you have "." on your PATH.  You should never do this on a linux system, as it is a potential  security risk.  It also breaks gcc builds.  The cross compiler you are building needs a cross assembler, so a link (or shell script) is put in the gcc build dir.  If you have "." on your PATH, then the native compiler will accidentally invoke the cross-assembler, and you will get odd errors.  You can also get the same problem if you have an empty string in your PATH list, or if you PATH starts or ends with a colon ':'.

There might also be something else wrong here, as you got a shell error instead of an assembler error.  It isn't obvious why that happened.

mayank2016, 2019-05-22:
now i am getting this error
-----------------------------------------------------
----------------------------------------------------
TARGET_CPU_DEFAULT="" \
HEADERS="config/riscv/riscv-protos.h config/linux-protos.h tm-preds.h" DEFINES="" \
/bin/sh ../.././riscv-gcc/gcc/mkconfig.sh tm_p.h
TARGET_CPU_DEFAULT="" \
HEADERS="auto-host.h ansidecl.h" DEFINES="" \
/bin/sh ../.././riscv-gcc/gcc/mkconfig.sh bconfig.h
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings -fno-common  -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/build -I../.././riscv-gcc/gcc/../include  -I../.././riscv-gcc/gcc/../libcpp/include  \
	-o build/genmodes.o ../.././riscv-gcc/gcc/genmodes.c
**/opt/riscv/riscv64-unknown-linux-gnu/bin/as: unrecognized option '--64'
make[2]: *** [build/genmodes.o] Error 1
make[2]: Leaving directory `/localdata/mayank/again_do/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc'
make[1]: *** [all-gcc] Error 2**
make[1]: Leaving directory `/localdata/mayank/again_do/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [stamps/build-gcc-linux-stage1] Error 2

thanks & regards
Mayank

jim-wilson, 2019-05-22:
The dir /opt/riscv/riscv64-unknown-linux-gnu/bin/ should not be on your path.  The cross tools are in /opt/riscv/bin, that is the only dir you should add to your path, and you don't have to do that until after the tools are built.  The stuff in /opt/riscv/riscv64-unknown-linux-gnu/bin are subprograms for the tools to call, not programs that the user should be using.

mayank2016, 2019-05-22:
Thank you for reply.
after command make linux successfully 

when i tried to convert c file in to .s file
i got error:
root@mayank44:/home/mayank/riscv-gnu-toolchain# riscv64-unknown-linux-gnu-gcc loop.c -S 
riscv64-unknown-linux-gnu-gcc: command not found

thanks and regards
mayank

jim-wilson, 2019-05-22:
Did you put the install bin dir on your path?  What does "which riscv64-unknown-linux-gnu-gcc" say?  And check "echo $PATH" to make sure it is correct.

closed 2021-03-03 by lazyparser