riscv-collab/riscv-gnu-toolchain #448: recipe for target 'arch/riscv/kernel' failed

ItachiUchiha1998, 2019-05-29:
`/home/vinayak/RISCV/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: -shared not supported
collect2: error: ld returned 1 exit status
  LD      arch/riscv/kernel/vdso/vdso-syms.o
  OBJCOPY arch/riscv/kernel/vdso/vdso.so
/home/vinayak/RISCV/bin/riscv64-unknown-elf-objcopy: 'arch/riscv/kernel/vdso/vdso.so.dbg': No such file
arch/riscv/kernel/vdso/Makefile:51: recipe for target 'arch/riscv/kernel/vdso/vdso.so' failed
make[2]: *** [arch/riscv/kernel/vdso/vdso.so] Error 1
scripts/Makefile.build:489: recipe for target 'arch/riscv/kernel/vdso' failed
make[1]: *** [arch/riscv/kernel/vdso] Error 2
Makefile:1071: recipe for target 'arch/riscv/kernel' failed
make: *** [arch/riscv/kernel] Error 2
make: *** Waiting for unfinished jobs....
`
getting the following error while running the following command : `$ make ARCH=riscv CROSS_COMPILE=/home/vinayak/RISCV/bin/riscv64-unknown-elf- -j4`

jim-wilson, 2019-05-29:
You are trying to use an embedded elf compiler to build a linux kernel.  You need to use a linux compiler to build the linux kernel.

ItachiUchiha1998, 2019-05-29:
whats the command  for using linux kernel?

jim-wilson, 2019-05-30:
To build a linux compiler from riscv-gnu-toolchain, you can use the configure option --enable-linux.

ItachiUchiha1998, 2019-05-30:
It worked
thanks for help.

closed 2019-05-30 by ItachiUchiha1998

marcopeereboom, 2019-05-31:
Or you can edit the file and disable shared but don't listen to me. @jim-wilson is right!

mathmax12, 2019-09-20:
@jim-wilson 

Hi Jim,
I got the same error:
`make -C linux-5.1.3-lowrisc ARCH=riscv CROSS_COMPILE=/opt/riscv/bin/riscv64-unknown-elf- CONFIG_SERIAL_8250_CONSOLE=y CONFIG_VT_CONSOLE=n CONFIG_LOWRISC_VGA_CONSOLE=n -j 4
make[1]: Entering directory '/home/riscv_project/lowrisc-chip_ariane0d7/lowrisc-chip4/linux-5.1.3-lowrisc'
HOSTCC scripts/basic/fixdep
HOSTCC scripts/kconfig/conf.o
...
...
VDSOLD arch/riscv/kernel/vdso/vdso-dummy.o
CC init/init_task.o
CC arch/riscv/kernel/cpu.o
VDSOLD arch/riscv/kernel/vdso/vdso.so.dbg
/opt/riscv/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: -shared not supported
collect2: error: ld returned 1 exit status
LD arch/riscv/kernel/vdso/vdso-syms.o
OBJCOPY arch/riscv/kernel/vdso/vdso.so
/opt/riscv/bin/riscv64-unknown-elf-objcopy: 'arch/riscv/kernel/vdso/vdso.so.dbg': No such file
arch/riscv/kernel/vdso/Makefile:50: recipe for target 'arch/riscv/kernel/vdso/vdso.so' failed
make[3]: *** [arch/riscv/kernel/vdso/vdso.so] Error 1
scripts/Makefile.build:486: recipe for target 'arch/riscv/kernel/vdso' failed
make[2]: *** [arch/riscv/kernel/vdso] Error 2
make[2]: *** Waiting for unfinished jobs....
CC init/version.o
Makefile:1051: recipe for target 'arch/riscv/kernel' failed
make[1]: *** [arch/riscv/kernel] Error 2
make[1]: *** Waiting for unfinished jobs....
AR init/built-in.a
make[2]: warning: Clock skew detected. Your build may be incomplete.
make[1]: *** wait: No child processes. Stop.
Makefile:48: recipe for target 'riscv-pk/serial/vmlinux-serial' failed
make: *** [riscv-pk/serial/vmlinux-serial] Error 2`

I also tried the configure option --enable-linux when building the riscv-gnu-toolchain.
But this error still persists.

jim-wilson, 2019-09-20:
Your first line says
```
CROSS_COMPILE=/opt/riscv/bin/riscv64-unknown-elf-
```
which is an embedded elf toolchain.  That should be riscv64-unknown-linux-gnu- instead to use a linux toolchain.

mathmax12, 2019-09-20:
Thank you, Jim.
This resolves my issue.