riscv-collab/riscv-gnu-toolchain #452: gdb is broken

micnaelyuanfeng, 2019-06-04:
Latest of this branch gdb cannot build

E:\msys64\opt\riscv\bin\riscv64-unknown-elf-ar.exe: creating libnosys.a
E:\msys64\opt\riscv\bin\riscv64-unknown-elf-ar.exe: creating libnosys.a
make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
*** This configuration is not supported in the following subdirectories:
     zlib target-libquadmath target-libatomic target-libitm target-libsanitizer target-libvtv target-libmpx gnattools gotools target-libada target-libhsail-rt target-libgfortran target-libbacktrace target-libgo target-libffi target-libobjc target-libgomp target-liboffloadmic target-libssp
    (Any other directories should still work fine.)
configure: WARNING: decimal float is not supported for this target, ignored
E:\msys64\mingw32\bin\objcopy.exe: 'E:/msys64/usr/bin/ls': No such file
make[2]: *** [Makefile:2566: configure-build-libiberty] Error 1
make[2]: *** [Makefile:7866: configure-libiberty] Error 1
make[1]: *** [Makefile:907: all] Interrupt
make: *** [Makefile:536: stamps/build-gcc-newlib-stage2] Interrupt
make[2]: *** [Makefile:9104: configure-gdb] Error 1
make[1]: *** [Makefile:857: all] Interrupt

jim-wilson, 2019-06-04:
Windows builds are always tricky.  Build on linux instead if you can.  I don't have access to a windows machine in the office, so I can only provide limited help.

It isn't clear what is wrong here.  Try a build with -j1 to get a clearer error message.  It does look like the build is failing in gdb configure, which means you will have to look at the config.log file in the gdb build dir to figure out what the actual error is.

Have you managed to build previous versions on Windows?  If this has never built for you, then you may not have the right setup.  If this has built for you in the past, then maybe there is some new requirement that you are missing.

micnaelyuanfeng, 2019-06-04:
Hi I am trying to compile it on windows platform which is required....... Sorry

micnaelyuanfeng, 2019-06-04:
I build successfully just once last Monday, and then I tried another couple of times all failed

micnaelyuanfeng, 2019-06-04:
Can you help me to take a look at the config.log?

micnaelyuanfeng, 2019-06-04:
In file included from E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/exceptions.h:23,
                 from E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/utils.h:24,
                 from E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/defs.h:670,
                 from E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/record-btrace.c:22:
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/ui-out.h: In function 'void btrace_insn_history(ui_out*, const btrace_thread_info*, const btrace_insn_iterator*, const btrace_insn_iterator*, gdb_disassembly_flags)':
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/ui-out.h:214:18: warning: 'asm_list.ui_out_emit_type<ui_out_type_list>::m_uiout' may be used uninitialized in this function [-Wmaybe-uninitialized]
  214 |     m_uiout->end (Type);
      |     ~~~~~~~~~~~~~^~~~~~
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/record-btrace.c:786:35: note: 'asm_list.ui_out_emit_type<ui_out_type_list>::m_uiout' was declared here
  786 |   gdb::optional<ui_out_emit_list> asm_list;
      |                                   ^~~~~~~~
/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/rust-exp.y:23.1-12: warning: POSIX Yacc does not support %pure-parser [-Wyacc]
   23 | %pure-parser
      | ^~~~~~~~~~~~
/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/rust-exp.y:23.1-12: warning: deprecated directive, use ‘%define api.pure’ [-Wdeprecated]
   23 | %pure-parser
      | ^~~~~~~~~~~~
/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/rust-exp.y:28.1-7: warning: POSIX Yacc does not support %expect [-Wyacc]
   28 | %expect 1
      | ^~~~~~~
/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/rust-exp.y: warning: fix-its can be applied.  Rerun with option '--update'. [-Wother]
In file included from E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c:24:
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-data.h:42:3: error: 'WINDOW' does not name a type
   42 |   WINDOW *handle;     /* Window handle.  */
      |   ^~~~~~
In file included from E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c:25:
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-win.h:40:8: error: 'chtype' does not name a type; did you mean 'type'?
   40 | extern chtype tui_border_ulcorner;
      |        ^~~~~~
      |        type
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-win.h:41:8: error: 'chtype' does not name a type; did you mean 'type'?
   41 | extern chtype tui_border_urcorner;
      |        ^~~~~~
      |        type
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-win.h:42:8: error: 'chtype' does not name a type; did you mean 'type'?
   42 | extern chtype tui_border_lrcorner;
      |        ^~~~~~
      |        type
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-win.h:43:8: error: 'chtype' does not name a type; did you mean 'type'?
   43 | extern chtype tui_border_llcorner;
      |        ^~~~~~
      |        type
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-win.h:44:8: error: 'chtype' does not name a type; did you mean 'type'?
   44 | extern chtype tui_border_vline;
      |        ^~~~~~
      |        type
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-win.h:45:8: error: 'chtype' does not name a type; did you mean 'type'?
   45 | extern chtype tui_border_hline;
      |        ^~~~~~
      |        type
In file included from E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c:26:
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-io.h:31:37: error: 'WINDOW' has not been declared
   31 | extern void tui_puts (const char *, WINDOW * = nullptr);
      |                                     ^~~~~~
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-io.h:52:35: error: variable or field 'tui_set_reverse_mode' declared void
   52 | extern void tui_set_reverse_mode (WINDOW *w, bool reverse);
      |                                   ^~~~~~
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-io.h:52:35: error: 'WINDOW' was not declared in this scope
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-io.h:52:43: error: 'w' was not declared in this scope
   52 | extern void tui_set_reverse_mode (WINDOW *w, bool reverse);
      |                                           ^
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-io.h:52:46: error: expected primary-expression before 'bool'
   52 | extern void tui_set_reverse_mode (WINDOW *w, bool reverse);
      |                                              ^~~~
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c: In function 'unsigned int tui_dispatch_ctrl_char(unsigned int)':
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c:59:10: error: 'KEY_NPAGE' was not declared in this scope
   59 |     case KEY_NPAGE:
      |          ^~~~~~~~~
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c:62:10: error: 'KEY_PPAGE' was not declared in this scope
   62 |     case KEY_PPAGE:
[... 11 lines cut ...]
   70 |     case KEY_SR:
      |          ^~~~~~
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c:73:10: error: 'KEY_RIGHT' was not declared in this scope
   73 |     case KEY_RIGHT:
      |          ^~~~~~~~~
[... cut ...]
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c:97:36: error: 'struct tui_gen_win_info' has no member named 'handle'
   97 |   WINDOW *w = TUI_CMD_WIN->generic.handle;
      |                                    ^~~~~~
E:/msys64/home/yfeng1/riscv-gnu-toolchain/riscv-gdb/gdb/tui/tui-command.c:99:3: error: 'wrefresh' was not declared in this scope
   99 |   wrefresh (w);
      |   ^~~~~~~~
make[3]: *** [Makefile:1637: tui/tui-command.o] Error 1
make[2]: *** [Makefile:9134: all-gdb] Error 2
make[1]: *** [Makefile:857: all] Error 2
make: *** [Makefile:420: stamps/build-gdb-newlib] Error 2

micnaelyuanfeng, 2019-06-04:
This is what I got from -j1

jim-wilson, 2019-06-04:
The WINDOWS type should be defined in a [n]curses.h header file.  That isn't a commonly used header file anymore, so you might not have that msys/mingw package installed.  Alternatively, you can add --disable-tui to the gdb configure options to turn this feature off.  tui is a terminal based user interface for gdb, for people that want a gui, but don't have a graphical display.  It isn't very useful.

Looks like a possible gdb bug that it isn't giving a better error message when there is no [n]curses package installed.

If you look at the config.h file in the gdb build dir, one of the HAVE_*CURSES* macros should be defined if you have one of the various curses packages installed.  If none of them are defined then you will need to install one of the curses packages.

hyf6661669, 2019-06-25:
Have you tried using cygwin?

amdfernlund, 2020-04-17:
In order to get riscv64-unknown-elf-gdb.exe to build under mingw/msys64 I had to address these two issues:
http://sourceware-org.1504.n7.nabble.com/PATCH-Make-GDB-compile-with-Python-3-on-MinGW-td585090.html
https://github.com/msys2/MINGW-packages/issues/5803

closed 2021-03-08 by lazyparser