riscv-collab/riscv-gnu-toolchain #453: How do i get a trace of qemu, which emmits the only the

fkladdu, 2019-06-07:


fkladdu, 2019-06-07:
Hi Every one,
       I have used the spike to get the executing code snippet, it emitted but it was emitting the whole code as same as assemble file. For this sake i tried to use the qemu but unable to get it. so can anyone Help me to get the Actual executing code snippet trace, which i have to pass as an input to the profiler?
         
            To get the trace of an executable from spike i have used the following command,

$ spike -I pk a.out 2>log

From this command, i got the trace file But it emitted the whole assemble file, Not the only executing instructions, i wat Actual Executing instructions to emit by spike or qemu.

Thanks,
khan

jim-wilson, 2019-06-07:
spike -l does emit a trace of the instructions that are executing.  Not clear why you think it doesn't.  If you look at branches in the trace file, you will see that the trace is following taken branches.

Perhaps you are confused by the fact that you are using pk?  Spike is a bare metal simulator, so you are actually emulating the pk program which in turn is emulating your a.out program.  When I try this with a simple hello world program, pk is the first 91% of the trace file,  You can tell the difference between the pk and a.out code because the pk code has address 0x000000008... and the a.out code has address 0x000000000... 

qemu isn't very useful for instruction traces, as it does jit compilation.

fkladdu, 2019-06-12:
Hi Jim & Every one,

--> what u said was right on the spike, it's emitting the trace of the executing instructions with pk instructions.
--> This think is working on the bare metal code. But due to Spike is a bare metal simulator, am not able Run the the Linux executable on spike, if Run the code on the spike then am getting a segfault as follows,

$ riscv64-unknown-linux-gnu-gcc hello.c 
$ spike pk a.out
bbl loader
z  0000000000000000 ra 00000000000103d4 sp 000000007f7e9b50 gp 0000000000012800
tp 0000000000000000 t0 0000000000000000 t1 0000000000000008 t2 0000000000012390
s0 0000000000000000 s1 0000000000000000 a0 00000000000104b6 a1 0000000000000001
a2 000000007f7e9b58 a3 00000000000104d8 a4 0000000000010530 a5 0000000000000000
a6 000000007f7e9b50 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 ffffffffffffffff t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc fffffffffffffffe va fffffffffffffffe insn       ffffffff sr 8000000200046020
User fetch segfault @ 0xfffffffffffffffe

--> Now i want the trace(hello world code compiled with the riscv-unknown-linux) of Linux platform instead of bare metal, 
--> is there any Linux Based simulator? 
--> if there, let me know, it will be soo Helpful to me.
                               
Thanks & Regards,
khan.

aswaterman, 2019-06-12:
Try `riscv64-unknown-linux-gnu-gcc -static hello.c`.  By default, the Linux compiler uses dynamic linking, which is not supported by `pk`.

fkladdu, 2019-06-12:
Thank you very much it works.

cghalibkhan, 2019-06-14:
Hi 
--  I just want to know, how qemu options are used to emit the trace of the app.
--  I have used these options But still not able to keep them in order.
-- Run the following command.

$ qemu-riscv64 -trace [[enable=]<pattern>][,events=<file>][,file=<file>]

Help me to get the trace from qemu-riscv64.

Thanks.

jim-wilson, 2019-06-14:
qemu is a jit.  You can't get good traces from it.

cghalibkhan, 2019-06-18:
Hi 
    To get Trace file, i was using a simulator gem5.
Build a RISCV configured simulator, &  
Run a simple Hello world code. 
$ ./build/RISCV/gem5.opt  --debug-flags=ExecUser ./configs/example/se.py a.out

Got an Error:
gem5 Simulator System.  http://gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 compiled Jun 17 2019 18:51:36
gem5 started Jun 18 2019 18:29:56
gem5 executing on exaleapserver-Inspiron-3470, pid 15617
command line: ./build/RISCV/gem5.opt --debug-flags=ExecUser ./configs/example/se.py a.out

Error: script doesn't take any positional arguments

Please, anyone, Help me to Get a Trace file From gem5.

Regards,
khan.

kito-cheng, 2019-06-19:
gdb simulator might be a good candidate for getting trace, the only limitation is it only support ELF target.

```
$ riscv32-unknown-elf-gcc hello.c
$ riscv32-unknown-elf-run --trace a.out
```

kito-cheng, 2019-06-19:
It might be what you want, single step mode plus dumping target assembly code.
`qemu-riscv32 --singlestep  -d in_asm`

cghalibkhan, 2019-06-20:
Thank u for the riscv run option, But still, it doesn't work for me because, I want a trace of Linux system, then, in that case, it is not that much usefull to me...
And coming to qemu option wat u have mentioned, it also not emitting a trace of exe, i used this options 

$ qemu-riscv64 -singlestep -d trace:help a.out 

Here i got so many options, I tried many options But didn't get any trace, it's just executing it & emitting output.
 
Help me to get Linux system trace of executable with qemu or gem5 simulators.

Thanks 
Ghalib khan.

closed 2021-03-08 by lazyparser