riscv-collab/riscv-gnu-toolchain #454: How to Build Cross Native Toolchain

cghalibkhan, 2019-06-13:
Hi Every one,

I tried to Build a Riscv cross Native Toolchain with these options, as follows.

$ ../configure --with-host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu --prefix=$(PWD)/riscv-gnu-toolchain/Exe --without-system-zlib

$ make linux

Error:
checking build system compiler gcc... no
configure: error: Specified CC_FOR_BUILD doesn't seem to work
Makefile:4669: recipe for target 'configure-gmp' failed
make[2]: *** [configure-gmp] Error 1
make[2]: Leaving directory '/home/riscv-gnu-toolchain/Build/build-gcc-linux-stage1'
Makefile:276: recipe for target 'stamps/build-gcc-linux-stage1' failed
make[1]: *** [stamps/build-gcc-linux-stage1] Error 2
make[1]: Leaving directory '/home/riscv-gnu-toolchain/Build'

Help me to Build Cross Native Toolchain...

Thanks & Regards,
khan.

jim-wilson, 2019-06-13:
1) Use a real distro like Fedora/Debian/OpenSuse/OpenEmbedded that comes with a native compiler.  Basically, anything other than whatever unspecified distro you are using.
2) Try the info at
    https://gcc.gnu.org/ml/gcc-help/2019-06/msg00042.html
   This assumes that you already have a C library on the target.
3) Try using crosstool-ng if you don't already have a C library on the target.

Building a toolchain this way is complicated.  If you really want to do it, you should expect to spend at least a few weeks learning how to do it.  It is much easier to use a toolchain that someone else built if you don't want to become a toolchain building expert.

jim-wilson, 2019-06-13:
You can't build a cross native compiler this way.  There is no good way to build a cross native compiler using riscv-gnu-toolchain.  It was never designed for that.

closed 2021-03-07 by lazyparser