riscv-collab/riscv-gnu-toolchain #455: Failed building 32bit Newlib

weedmank, 2019-06-14:
**Used the following to make the Linux version and all went well:**
./configure --prefix=/opt/riscv --with-arch=rv32gc
make linux

**However it didn't work for a Newlib build**
./configure --prefix=/opt/riscv --with-arch=rv32gc
make

**I got the following failure:**
...
...
 for i in argz/lib.a stdlib/lib.a ctype/lib.a search/lib.a stdio/lib.a   string/lib.a signal/lib.a time/lib.a locale/lib.a reent/lib.a  errno/lib.a misc/lib.a ssp/lib.a   syscalls/lib.a  machine/lib.a ; do \
   riscv32-unknown-elf-ar x ../$i; \
 done; \
riscv32-unknown-elf-ar rc ../libc.a *.o
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 2: riscv32-unknown-elf-ar: command not found
/bin/bash: line 4: riscv32-unknown-elf-ar: command not found
Makefile:1031: recipe for target 'libc.a' failed
make[6]: *** [libc.a] Error 127
make[6]: Leaving directory '/home/kirk/riscv/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/newlib/libc'
Makefile:683: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/kirk/riscv/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/newlib/libc'
Makefile:641: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/kirk/riscv/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/newlib'
Makefile:452: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/kirk/riscv/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/newlib'
Makefile:8491: recipe for target 'all-target-newlib' failed
make[2]: *** [all-target-newlib] Error 2
make[2]: Leaving directory '/home/kirk/riscv/riscv-gnu-toolchain/build-newlib'
Makefile:878: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/kirk/riscv/riscv-gnu-toolchain/build-newlib'
Makefile:470: recipe for target 'stamps/build-newlib' failed
make: *** [stamps/build-newlib] Error 2

**Here's what's in my /opt/riscv/bin folder after failure:**
kirk@Ubuntu16:~/riscv/riscv-gnu-toolchain$ ls /opt/riscv/bin/
riscv32-unknown-elf-c++         riscv32-unknown-linux-gnu-addr2line  riscv32-unknown-linux-gnu-gcc-nm         riscv32-unknown-linux-gnu-nm
riscv32-unknown-elf-cpp         riscv32-unknown-linux-gnu-ar         riscv32-unknown-linux-gnu-gcc-ranlib     riscv32-unknown-linux-gnu-objcopy
riscv32-unknown-elf-g++         riscv32-unknown-linux-gnu-as         riscv32-unknown-linux-gnu-gcov           riscv32-unknown-linux-gnu-objdump
riscv32-unknown-elf-gcc         riscv32-unknown-linux-gnu-c++        riscv32-unknown-linux-gnu-gcov-dump      riscv32-unknown-linux-gnu-ranlib
riscv32-unknown-elf-gcc-8.3.0   riscv32-unknown-linux-gnu-c++filt    riscv32-unknown-linux-gnu-gcov-tool      riscv32-unknown-linux-gnu-readelf
riscv32-unknown-elf-gcc-ar      riscv32-unknown-linux-gnu-cpp        riscv32-unknown-linux-gnu-gdb            riscv32-unknown-linux-gnu-run
riscv32-unknown-elf-gcc-nm      riscv32-unknown-linux-gnu-elfedit    riscv32-unknown-linux-gnu-gdb-add-index  riscv32-unknown-linux-gnu-size
riscv32-unknown-elf-gcc-ranlib  riscv32-unknown-linux-gnu-g++        riscv32-unknown-linux-gnu-gfortran       riscv32-unknown-linux-gnu-strings
riscv32-unknown-elf-gcov        riscv32-unknown-linux-gnu-gcc        riscv32-unknown-linux-gnu-gprof          riscv32-unknown-linux-gnu-strip
riscv32-unknown-elf-gcov-dump   riscv32-unknown-linux-gnu-gcc-8.3.0  riscv32-unknown-linux-gnu-ld
riscv32-unknown-elf-gcov-tool   riscv32-unknown-linux-gnu-gcc-ar     riscv32-unknown-linux-gnu-ld.bfd

jim-wilson, 2019-06-14:
That looks very odd.  Your embedded elf newlib build failed because binutils did not install properly.  But your embedded elf gcc did build and install which could not have worked if binutils didn't build and at least partially install.

I would guess that something happened during the build causing it to fail, you tried to fix it and continue, but that left the build and/or install tree in an inconsistent state preventing the build from completing.  If this is the case, then rebuilding from scratch should work.  You might be able to fix this with "cd build-binuitils-newlib; make install", but it is safer to rebuild from scratch since if one thing is broken there might also be other hidden breakage.

A second guess is that you accidentally corrupted the install tree while the build was in progress.

It might also help to have a clean install tree.  Trying to install a new release over top of an old release can sometimes cause problems, but I don't think that it could give this result.

weedmank, 2019-06-14:
Looks like that was the problem....I must have accidentally corrupted it. I removed the /opt/riscv folder, created a new one as well as removing the /home/kirk/riscv/riscv-gnu-toolchain folder.  I then re-cloned the riscv-gnu-toolchain and did it again and all is well.  Thank you Jim!

closed 2019-06-14 by weedmank