riscv-collab/riscv-gnu-toolchain #457: Compile Error

flylcl, 2019-06-24:
I got this error message after run compile in eclipse.
Hope someone help me.
Thanks.
[100%] Linking C executable ../../../core.bin
cd /home/flylcl/Project/EN675/Validation/en675_fpga_rc002_190624094556/Debug/Source/core/shared && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/core.bin.dir/link.txt --verbose=1
/opt/riscv1/bin/riscv64-unknown-elf-gcc -DDEBUG -DC_FLAGS -O0 -MMD -MP -Wall -D__NO_INLINE__ -mcmodel=medany -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -g  -static -Wl,--gc-sections -T /home/flylcl/Project/EN675/Validation/en675_fpga_rc002_190624094556/Source/core/shared/exbl.lds -DDEBUG -DLINKER_FLAGS -Wl,--build-id=none -nostartfiles -static CMakeFiles/core.bin.dir/boot.c.obj CMakeFiles/core.bin.dir/exbl.c.obj CMakeFiles/core.bin.dir/msg.c.obj CMakeFiles/core.bin.dir/user.c.obj CMakeFiles/core.bin.dir/mentry.S.obj  -o ../../../core.bin  -L/home/flylcl/Project/EN675/Validation/en675_fpga_rc002_190624094556/Debug/lib ../../../lib/libcore0.a ../../../lib/libcore1.a ../../../lib/libcore2.a ../../../lib/libcore3.a ../../../lib/libfatfs.a ../../../lib/liblwip.a ../../../lib/libfreertos.a ../../../lib/libdev.a ../../../lib/libenxisp.a 
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/libgcc.a(_clzsi2.o): in function `__clzdi2':
/home/flylcl/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv64-unknown-elf/libgcc/../../.././riscv-gcc/libgcc/libgcc2.c:710:(.text+0x1c): relocation truncated to fit: R_RISCV_HI20 against symbol `__clz_tab' defined in .rodata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/libgcc.a(_clz.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fflush.o): in function `.L48':
fflush.c:(.text+0x198): relocation truncated to fit: R_RISCV_HI20 against symbol `_global_impure_ptr' defined in .srodata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o): in function `std':
findfp.c:(.text+0x38): relocation truncated to fit: R_RISCV_HI20 against symbol `__sread' defined in .text section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-stdio.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-freer.o): in function `_malloc_trim_r':
mallocr.c:(.text+0x4): relocation truncated to fit: R_RISCV_HI20 against symbol `__malloc_av_' defined in .data section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mallocr.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-getenv.o): in function `_findenv':
getenv.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-getenv_r.o): in function `_findenv_r':
getenv_r.c:(.text+0x1e): relocation truncated to fit: R_RISCV_HI20 against symbol `environ' defined in .sdata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-environ.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-lcltime_r.o): in function `.L3':
lcltime_r.c:(.text+0x50): relocation truncated to fit: R_RISCV_HI20 against symbol `__month_lengths' defined in .rodata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-month_lengths.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-locale.o): in function `_setlocale_r':
locale.c:(.text+0x8): relocation truncated to fit: R_RISCV_HI20 against `.LC0'
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mallocr.o): in function `.L2':
Source/core/shared/CMakeFiles/core.bin.dir/build.make:144: recipe for target 'core.bin' failed
gmake[2]: Leaving directory '/home/flylcl/Project/EN675/Validation/en675_fpga_rc002_190624094556/Debug'
CMakeFiles/Makefile2:714: recipe for target 'Source/core/shared/CMakeFiles/core.bin.dir/all' failed
gmake[1]: Leaving directory '/home/flylcl/Project/EN675/Validation/en675_fpga_rc002_190624094556/Debug'
Makefile:105: recipe for target 'all' failed
mallocr.c:(.text+0x62): relocation truncated to fit: R_RISCV_HI20 against symbol `__malloc_av_' defined in .data section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mallocr.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mbtowc_r.o): in function `_mbtowc_r':
mbtowc_r.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against symbol `_impure_ptr' defined in .sdata section in /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv1/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-mktime.o): in function `.L15':
mktime.c:(.text+0xb6): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
gmake[2]: *** [core.bin] Error 1
gmake[1]: *** [Source/core/shared/CMakeFiles/core.bin.dir/all] Error 2
gmake: *** [all] Error 2
"/usr/bin/gmake -j4 all" terminated with exit code 2. Build might be incomplete.

jim-wilson, 2019-06-24:
Is this a compiler that you built yourself, or one that you downloaded from somewhere?

I would guess that the libraries were not compiled -mcmodel=medany, and you are linking to an address at or above 0x80000000.  If this is a compiler you built yourself, then add --with-cmodel=medany when configuring.

kito-cheng, 2020-10-12:
Close this issue, long time no further discussion/response.

closed 2020-10-12 by kito-cheng