riscv-collab/riscv-gnu-toolchain #459: How to build riscv64-linux-gnu-gccgo and riscv64-unknown-elf-gccgo

mohanson, 2019-06-26:
I searched your code globally and found that there are `gccgo` related configurations, but I tried it many times but still can't build it normally. Since it takes a long time to compile once, please allow this question to be raised here: 

*How could I build `riscv64-linux-gnu-gccgo`  and `riscv64-unknown-elf-gccgo` with this project?*

Thanks.

jim-wilson, 2019-06-26:
Add ",go" to the the build-gcc-linux-stage2 --enable-languages configure option.  However, I don't know if this will work with the gcc-8 that we have in riscv-gnu-toolchain.  You might need gcc-9 to make this work.  I think RISC-V go support is a relatively recent addition.

mohanson, 2019-06-27:
I see the log showing that it's compiling `gccgo`, but something went wrong:

```
In file included from /opt/riscv/riscv32-unknown-elf/include/dirent.h:39,
                 from ../../../../riscv-gcc/libgo/sysinfo.c:14:
/opt/riscv/riscv32-unknown-elf/include/sys/dirent.h:10:2: error: #error "<dirent.h> not supported"
 #error "<dirent.h> not supported"
  ^~~~~
In file included from ../../../../riscv-gcc/libgo/sysinfo.c:14:
/opt/riscv/riscv32-unknown-elf/include/dirent.h:48:12: error: unknown type name 'DIR'
 int  dirfd(DIR *);
            ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:51:17: error: unknown type name 'DIR'
 int  fdclosedir(DIR *);
                 ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:53:1: error: unknown type name 'DIR'
 DIR *opendir(const char *);
 ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:54:1: error: unknown type name 'DIR'
 DIR *fdopendir(int);
 ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:56:11: error: unknown type name 'DIR'
   readdir(DIR *);
           ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:58:16: error: unknown type name 'DIR'
 int  readdir_r(DIR *__restrict, struct dirent *__restrict,
                ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:61:17: error: unknown type name 'DIR'
 void  rewinddir(DIR *);
                 ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:72:15: error: unknown type name 'DIR'
 void  seekdir(DIR *, long);
               ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:73:15: error: unknown type name 'DIR'
 long  telldir(DIR *);
               ^~~
/opt/riscv/riscv32-unknown-elf/include/dirent.h:76:15: error: unknown type name 'DIR'
 int  closedir(DIR *);
               ^~~
../../../../riscv-gcc/libgo/sysinfo.c:17:10: fatal error: ucontext.h: No such file or directory
 #include <ucontext.h>
          ^~~~~~~~~~~~
compilation terminated.
```

I'm sure I have `ucontext.h` in my system

```
[root@ /src/riscv-gnu-toolchain]$ locate ucontext.h

/usr/include/ucontext.h
/usr/include/asm-generic/ucontext.h
/usr/include/sys/ucontext.h
/usr/include/x86_64-linux-gnu/asm/ucontext.h
/usr/include/x86_64-linux-gnu/sys/ucontext.h
/usr/include/x86_64-linux-musl/ucontext.h
/usr/include/x86_64-linux-musl/sys/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/linux-headers/include/asm/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/linux-headers/include/asm-generic/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-gcc/fixincludes/tests/base/sys/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/conform/data/ucontext.h-data
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/include/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/stdlib/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/sysdeps/arm/sys/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/sysdeps/generic/sys/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/sysdeps/i386/sys/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/sysdeps/m68k/sys/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/sysdeps/mips/sys/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/sysdeps/unix/sysv/linux/aarch64/sys/ucontext.h
/disk/sata1/src/riscv-gnu-toolchain/riscv-glibc/sysdeps/unix/sysv/linux/alpha/sys/ucontext.h
[... 16 lines cut ...]
/usr/riscv64-linux-gnu/include/asm/ucontext.h
/usr/riscv64-linux-gnu/include/asm-generic/ucontext.h
/usr/riscv64-linux-gnu/include/sys/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/alpha/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/arm/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/arm64/include/uapi/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/cris/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/frv/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/ia64/include/uapi/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/m32r/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/m68k/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/mips/include/uapi/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/mn10300/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/parisc/include/asm/compat_ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/parisc/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/powerpc/include/uapi/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/riscv/include/uapi/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/s390/include/uapi/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/score/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/x86/include/uapi/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/arch/xtensa/include/asm/ucontext.h
/usr/src/linux-headers-4.15.0-29/include/uapi/asm-generic/ucontext.h
```

Do I need to add some paths to INCLUDE_PATH? You see there are many files named `ucontext.h` in your project or my system, I'm wired

jim-wilson, 2019-07-04:
It isn't the x86_64-linux dirent.h that is the problem.  It is the riscv64-elf dirent.h that is the problem.  This makes sense, because you can't support filesystem syscalls when you have no filesystem and no OS.  Maybe gccgo doesn't support bare-metal targets, and you need to configure for a target with an OS instead, like riscv64-linux or riscv64-rtems.

mohanson, 2019-07-04:
I understand. At present, it is impossible to compile bare metal gccgo, I really tried it here for a long time. thanks!

closed 2019-07-04 by mohanson