riscv-collab/riscv-gnu-toolchain #460: Compile V-Extension Instructions

wyahiaoui, 2019-06-27:
I am actually working on the risc V ISA where I want to make optimization through the V-Extension. 

I wanted to develop a small application using the simd instruction where I built some simple vectors operation in the goal to get familiar with the instructions. I compiled using 
- riscv64-unkown-elf-gcc -march=rv64gcv -mabi=ilp64d -o hello_world hello_world.c

But I get unrecognized substring. I updated the toolchain many times, I deleted it and reintalled it again but it doesn't bring any result. 

When I remove the march and abi, I get the unrecognized opcode. 

Can u help me please to get it work?

Thanks

jim-wilson, 2019-06-27:
There is no compiler support for the vector extension as yet.

You can find binutils support in github.com/sifive/riscv-binutils-gdb on the rvv branch.  Gas does accept the -march=rv64gcv option if you use that branch.

I expect that eventually the binutils patches will be moved into this tree and token gcc patches added to accept the -march=rv64gcv option and pass it to the assembler.  Meanwhile, you have to use the sifive tree, and use -Wa,-march=rv64gcv if you want to use gcc with extended asms.

wyahiaoui, 2019-06-30:
Sir Wilson, 

I saw also your publication about porting the opcodes in binutils in google forum and it seems interesting. 

I made this steps but it doesn't bring any result with the vector instruction set:

- Git clone https://github.com/riscv/riscv-binutils-gdb.git
- git checkout rvv
- mkdir build && cd build
- export RISCV=/path/to/install 
- ../configure  --prefix=$RISCV
- make 
- make install 

I wanted just to compile a simple program which contains a vector instruction 

.global main
main:
       vsetvli t0, x0, e32
       vadd v0, v0, v0

I compiled using: 
- riscv64-linux-gnu-as -march=rv32gcv -mabi=lp32f vector.S -o vector.out 

it seems recognizing the v substring in march but it still saying unrecognized opcode. 

Do you know please whether I missed a step or I made a wrong one?

jim-wilson, 2019-07-01:
The riscv/riscv-gnu-toolchain rvv branch is actually the 0.6 vector extension support.  You want the 0.7 vector extension support.  This can be found in the sifive/riscv-binutils-gdb rvv branch.  It needs to be moved into riscv/riscv-gnu-toolchain but we have a naming problem, and I'm badly oversubscribed so I haven't gotten around to figuring out how to fix it yet.  Also, the V extension group has made yet another incompatible change and changed the version number, so we need to figure out how to handle that.  I don't know if anyone is still using the 0.6 vector support, so I don't know if I can safely overwrite that.  Maybe we just create a bunch of branches (or tags?) for rvv-0.7.x, rvv-0.7.1, rvv-0.7.2, etc., where rvv-0.7.x is the mainline that has the most recent versions and rvv-0.7.1 matches the v0.7.1 spec, etc.  That will take some time to do, and I don't have a lot of time to work on this.

The rvv 0.7.x  port has a testcase that includes every syntax variant it accepts, if you aren't sure about the syntax.  See for instance
https://github.com/sifive/riscv-binutils-gdb/blob/rvv/gas/testsuite/gas/riscv/vector-insns.s

jim-wilson, 2019-07-02:
I've created rvv-0.7.1 and rvv-0.7.x branches in riscv/riscv-gnu-toolchain, which for now are the same.  But I'm about to write a patch for the latest architecture change and then they will be different as that will only be added to the rvv-0.7.x branch.

I also added some trivial gcc support so that it can accept -march=rv64gcv and define __riscv_vector.

wyahiaoui, 2019-07-05:
@jim-wilson Sir, After you updated the repository, install it and test it. It works as magic. I am thankful for the great Work that you did. All my Respect. 

Issue can be closed.

closed 2019-07-05 by wyahiaoui

allenderalencar, 2019-10-07:
@wyahiaoui 
Hello, I tried your steps before and some variations and just get opcode problem. Can you tell the steps you did to get vector instructions working?

thank you so much.

kito-cheng, 2019-10-07:
Hi @allenderalencar: 
Did you have tried rvv-0.7.x branch? it should be implemented most latest spec for vector extension, feel free to report any opcode problem :)
```
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-0.7.x
cd riscv-gnu-toolchain-rvv
git submodule update --init --recursive
mkdir build && cd build
export RISCV=/path/to/install
../configure --prefix=$RISCV
make
make install
```

SGR0101, 2021-03-31:
> @wyahiaoui
> Hello, I tried your steps before and some variations and just get opcode problem. Can you tell the steps you did to get vector instructions working?
> 
> thank you so much.

Can you explain your steps...Because I'm facing "unregnized opcode" error.
Thank you