riscv-collab/riscv-gnu-toolchain #462: Generating code for load / store double word.

JoshSabet, 2019-07-03:
The insn_code that is generated for each ld/sd at the beginning of a procedure is from what I can tell generated in expr.c, emit_move_insn_1(rtx x, rtx y):

code = optab_handler (mov_optab, mode);

But optab_handler () calls raw_optab_handler()

From what I can tell raw_optab_handler() is defined in genopinit.c as a string. Is this correct?

jim-wilson, 2019-07-03:
As mentioned before, once you  call emit_move_insn, you have already decided what kind of instruction to emit.  If you want a different instruction, it has to be changed higher up.  The place where you should be looking is riscv_for_each_saved_reg where it calls riscv_save_restore_reg, as this is handling one register at a time.  You need to change the loop to handle two registers at a time.

raw_optab_handler is a function that is generated at build time.  You have to look at insn-opinit.c in a build tree to find it.  What you are looking at in genopinit.c is part of the code that generates the raw_optab_handler function at build time.

You don't need to understand how optabs work in order to generate a load/store double word instruction.  You just need to pass different rtx to the emit_move_insn function.  If you have a 32-bit target, and are generating code for prologues, then you will have (mem:SI ...) as x and (reg:SI ...) as y.  To get a store double word instead, you want to pass in (mem:DI ...) as x and (reg:DI ...) as y.  You also need a movdi pattern in the md file that can emit your load/store double word instruction.

JoshSabet, 2019-07-03:
To emit a different kind of instruction I would need to generate the opcode and funct3 of my instruction which saves each register specified by rd (a bit vector). 

I tried following the code from riscv_for_each_saved_reg ()

riscv_for_each_saved_reg () calls
riscv_save_restore_reg () which calls
riscv_save_reg () which calls
riscv_emit_move () which calls
emit_move_insn () which calls
emit_move_insn_1 ()

I thought the code of the instruction was generated in emit_move_insn_1 () as it contains the lines
"enum insn_code code;" and "emit_insn (GEN_FCN (code) (x, y));"

jim-wilson, 2019-07-04:
riscv_save_restore_reg is creating the memory rtl and the register rtl.  So we already know what instruction we want here.  It is only ending up in emit_move_insn at the end for convenience. to create the move instruction.

​optabs are primarily used by the target independent optimizer when it needs to emit an operation but can't know what operations are supported by the target.  We indirect through the optabs for this.  ​Optabs is a table of operations supported by the target.  ​But since we are in the riscv backend, we know what riscv operations are supported and don't need an optab here.

It is sometimes convenient to use emit_move_insn though instead of creating our own move instructions by hand.​  Move instructions have a number of special requirements, e.g. if it is a double word move you might need to split it into two word size moves and do some address arithmetic.  If you are moving memory to memory you might need to go through a temporary register.  Etc.  emit_move_insn knows how to deal with all of this stuff.  It will also do the housekeeping stuff for you, like inserting the instruction into the instruction stream in the right place.

But in this case, you want to do something very different, emit a load/store multiple via a mask.  So what you should do is add a special pattern to the riscv.md file, e.g. store_multiple_with_mask, and then in riscv_for_each_saved_reg you call gen_store_multiple_with_mask () passing in the load address and the mask as arguments.  You will need to generate rtl for those.  The load address will be the same as what is generated in the first riscv_save_restore_reg call.  And the load mask can just be GEN_INT (mask) where the mask will come from cfun->machine->frame.mask with the eh return regs filtered out.  See for instance the arm port which has load/store multiple.  This is emit_multi_reg_push I think.​  This function creates a parallel of every register being stored from the mask, so you might need to do something like this for riscv too.​  See for instance the patterns in the arm/ldmstm.md  file.

closed 2021-03-08 by lazyparser