riscv-collab/riscv-gnu-toolchain #463: error when using riscv64-unknown-elf-g++

wky9710, 2019-07-03:
I've finished making and got all the cross-compilers, but failed compiling a hello-world .cpp file.
Info:
$ riscv64-unknown-elf-g++ test.cpp
riscv64-unknown-elf-g++: error trying to exec 'cc1plus': execvp: No such file or directory

I can't find any info about 'cc1plus', how can I solve this problem?
------------------------------------
some says that this is caused by different versions of gcc and g++. My riscv cross-compilers are all gcc 8.3, and x86 g++ is also updated to 8.3. So maybe that's not the problem. Info below:

$ riscv64-unknown-elf-g++ -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-g++
Target: riscv64-unknown-elf
Configured with: /home/wky/work/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/wky/work/riscv-toolchain-install --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/wky/work/riscv-toolchain-install/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=.././riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 8.3.0 (GCC)

$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
Target: riscv64-unknown-elf
Configured with: /home/wky/work/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/wky/work/riscv-toolchain-install --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/wky/work/riscv-toolchain-install/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=.././riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 8.3.0 (GCC)

$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 8.3.0-6ubuntu1~18.04.1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 8.3.0 (Ubuntu 8.3.0-6ubuntu1~18.04.1)
-----------------------------------------------------
By the way, I came here because I failed building riscv/riscv-tools using default x86 compiler, and one issue there
https://github.com/riscv/riscv-tools/issues/285
is the exact error I met. Jim-wilson said I should build cross-compilers first for it does not contain compilers now. But specifying riscv64-unknown-elf-gcc still won't work for that, details below:

$ CC=riscv64-unknown-elf-gcc CXX=riscv64-unknown-elf-g++ ./build.sh
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
configure: error: in `/home/wky/work/riscv-tools/riscv-openocd/build':
configure: error: C compiler cannot create executables
See `config.log' for more details

Is the command line above correct?

Sorry I'm really new to this area, and thanks for your patience :)

wky9710, 2019-07-06:
My fault... It was because I used the relative path, so riscv64-unknown-elf-g++ didn't work...
By the way, no need of riscv-tools, just download gnu-toolchains, pk, spike seperately and you can run riscv.

closed 2019-07-06 by wky9710

Joey0538, 2021-04-12:
So can you please specify or provide screenshots on how to resolve this error as I'm facing the same issue. Please help sir. Been stuck on it for over a week.