riscv-collab/riscv-gnu-toolchain #465: Error in configuring for Berkeley bootloader

fsaud, 2019-07-07:
I am making the setup for riscv and want to run linux on riscv architecture. I am following the below guide.
[(https://buildmedia.readthedocs.org/media/pdf/risc-v-getting-started-guide/latest/risc-v-getting-started-guide.pdf]

I am following the chapter 8 of this guide.
I have followed the steps till building the linux kernel. While configuring the bbl, following log is being printed on the screen
fsaud@fsaud-VirtualBox:~/riscv/riscv-pk/build$ ../configure --enable-logo --host=riscv64-unknown-elf --with-payload=../../linux/vmlinux
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-elf
checking for riscv64-unknown-elf-gcc... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for riscv64-unknown-elf-g++... no
checking for riscv64-unknown-elf-c++... no
checking for riscv64-unknown-elf-gpp... no
checking for riscv64-unknown-elf-aCC... no
checking for riscv64-unknown-elf-CC... no
checking for riscv64-unknown-elf-cxx... no
checking for riscv64-unknown-elf-cc++... no
checking for riscv64-unknown-elf-cl.exe... no
checking for riscv64-unknown-elf-FCC... no
checking for riscv64-unknown-elf-KCC... no
checking for riscv64-unknown-elf-RCC... no
checking for riscv64-unknown-elf-xlC_r... no
checking for riscv64-unknown-elf-xlC... no
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for riscv64-unknown-elf-ar... no
checking for ar... ar
checking for riscv64-unknown-elf-ranlib... no
checking for ranlib... ranlib
checking for riscv64-unknown-elf-readelf... no
checking for readelf... readelf
checking for riscv64-unknown-elf-objcopy... no
checking for objcopy... objcopy
checking for a BSD-compatible install... /usr/bin/install -c
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
configure: configuring default subproject : pk
configure: configuring default subproject : bbl
configure: configuring default subproject : softfloat
configure: configuring default subproject : dummy_payload
configure: configuring default subproject : machine
configure: configuring default subproject : util
configure: creating ./config.status
config.status: creating pk.mk
config.status: creating bbl.mk
config.status: creating softfloat.mk
config.status: creating dummy_payload.mk
config.status: creating machine.mk
config.status: creating util.mk
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
Upon hitting the build command, I am getting the following error.

fsaud@fsaud-VirtualBox:~/riscv/riscv-pk/build$ make -j $(nproc)
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/file.c
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE    -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I../pk -I../bbl -I../softfloat -I../dummy_payload -I../machine -I../util -c ../pk/syscall.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
Makefile:319: recipe for target 'file.o' failed
make: *** [file.o] Error 1
make: *** Waiting for unfinished jobs....
Makefile:319: recipe for target 'syscall.o' failed
make: *** [syscall.o] Error 1

This seems to me that, while configuring the script is finding some binaries which it could not found. And the reason is that, they actually do not exist as per my understanding. 
Following are the files and folders present at the toolchain installation bin folder.
fsaud@fsaud-VirtualBox:~/riscv/riscv-pk/build$ ls ../../riscv-install/bin/
bin                                  riscv64-unknown-linux-gnu-ar       riscv64-unknown-linux-gnu-gcc-8.3.0   riscv64-unknown-linux-gnu-gcov-tool  riscv64-unknown-linux-gnu-objdump  share
include                              riscv64-unknown-linux-gnu-as       riscv64-unknown-linux-gnu-gcc-ar      riscv64-unknown-linux-gnu-gprof      riscv64-unknown-linux-gnu-ranlib   sysroot
lib                                  riscv64-unknown-linux-gnu-c++filt  riscv64-unknown-linux-gnu-gcc-nm      riscv64-unknown-linux-gnu-ld         riscv64-unknown-linux-gnu-readelf
libexec                              riscv64-unknown-linux-gnu-cpp      riscv64-unknown-linux-gnu-gcc-ranlib  riscv64-unknown-linux-gnu-ld.bfd     riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu            riscv64-unknown-linux-gnu-elfedit  riscv64-unknown-linux-gnu-gcov        riscv64-unknown-linux-gnu-nm         riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-addr2line  riscv64-unknown-linux-gnu-gcc      riscv64-unknown-linux-gnu-gcov-dump   riscv64-unknown-linux-gnu-objcopy    riscv64-unknown-linux-gnu-strip

Kindly help resolving this issue

jim-wilson, 2019-07-07:
You need a riscv64-unknown-elf-gcc to build bbl/pk.  You need a riscv64-unknown-linux-gnu-gcc to build the kernel.  So you need to build two compilers.

The error occurs because the riscv64 cross compiler it needed wasn't found, so it tried using a native x86_64 compiler, which doesn't accept riscv compiler options.

fsaud, 2019-07-08:
@jim-wilson from where I can get the riscv-unknown-elf-gcc. Is there any already built binaries present somwhere? Or if we have to build it manually, how would I do that? Is there any guide to build it? 
Thank you so much.

jim-wilson, 2019-07-08:
You can find pre-built binaries on sifive.com/boards and scroll down to the software section.  freedom studio is eclipse based tools and gnu embedded is command line tools.

You can build it from riscv/riscv-gnu-toolchain, --disable-linux (the default) gives an embedded elf toolchain and --enable-linux gives a linux toolchain, --enable-multilib gives a toolchain that suppors multiple arch/abi target, which is how SiFive builds the toolchain.  You can also build it from crosstool-ng if you enable experimental support.  Or from upstream FSF sources if you know how to do that.

On recent linux systems, you can just apt install to get a riscv64 linux toolchain.

fsaud, 2019-07-10:
Can I build baremetal application with the toolchain present on the sifive.com/boards?

jim-wilson, 2019-07-10:
> Can I build baremetal application with the toolchain present on the sifive.com/boards?

Yes.  The embedded elf toolchains there are built the same way that riscv/riscv-gnu-toolchains are built.  We also have freedom-e-sdk which is the BSP for our boards, to make it easier to compile bare metal applications for them.

fsaud, 2019-07-13:
How do we build riscv32 baremetal toolchain from riscv/riscv-gnu-toolchain? What are the flags?
I have configured it as:
./configure --prefix=/home/fsaud/riscv/riscv_elf_toolchain/risv32 --with-arch=rv32imafdc --with-abi=ilp32d

jim-wilson, 2019-07-13:
That will build a compiler that defaults to rv32gc/ilp32d and supports only that one arch/abi choice.

A common convention is to build a 64-bit multilib compiler that emits rv64gc/lp64d by default, but also supports about 18 different arch/abi choices, including rv32gc/ilp32d.  You only need --enable-multilib along with a --prefix= to get that.  This is how SiFive embedded cross compilers are built for instance.

The toolchain sources support building them from a directory outside the source tree.  This has multiple benefits.  You can build more than one toolchain from the same source tree.  Your source tree doesn't get confused by build artifacts.  Etc.  You just do something like
  mkdir build-rv32gc-ilp32d
  cd build-rv32gc-ilp32d
  ../riscv-gnu-toolchain/configure --with-arch=rv32gc --with-abi=ilp32d --prefix=...

fsaud, 2019-07-14:
What would be the make command for building the riscv32 baremetal toolchain or multilib toolchain. I have given the configure command with above mentioned options. After that when I gave make, it tells me that "Nothing to be done for all".

jim-wilson, 2019-07-14:
You apparently already built a toolchain.  if you think this is wrong, you can do "make clean" to remove the last toolchain you built, and then "make" will build a fresh one.

fsaud, 2019-07-15:
What's the difference in rv32imafdc and rv32gc?

jim-wilson, 2019-07-15:
rv32gc is easier to type.  Otherwise there is no difference.

kito-cheng, 2020-10-12:
Close this issue, seems issue resolved and long time no further discussion/response.

closed 2020-10-12 by kito-cheng