riscv-collab/riscv-gnu-toolchain #467: Can not find "riscv64-unknow-elf-gcc" in /opt/riscv/bin

JunnanLi, 2019-07-11:
Hi,

After run "make linux", I do not find "riscv64-unknow-elf-gcc" in /opt/riscv/bin (I have added PATH=/opt/riscv/bin:$PATH)

aswaterman, 2019-07-11:
The links compiler is named riscv64-unknown-linux-gnu-gcc, not riscv64-unknown-elf-gcc.

aswaterman, 2019-07-11:
Or, if you wanted to build riscv64-unknown-elf-gcc, you can try "make newlib" instead of "make linux".

Jesse-Millwood, 2019-07-30:
I did try 
```
../configure --prefix=/opt/x-tools --enable-multilib 
make newlib
```
And can confirm there is no `riscv64-unknown-elf-gcc` or other bin in the prefix bin directory.

jim-wilson, 2019-07-30:
We need a lot more info than that.  What exactly happened after "make newlib"?  Your problem is probably not related to the original problem.  New issues for new problems please.

Jesse-Millwood, 2019-07-30:
I will need to withdraw my comment. Mine installs just fine. sorry for the noise

closed 2021-03-03 by lazyparser