riscv-collab/riscv-gnu-toolchain #469: Issue softfloat compile

johnnavaj, 2019-07-14:
Hi,

I wanted to compile with softfloat , hence i added  -march=rv64imafd -mabi=lp64d with medany model and i am getting the below linking errors. 
[err.log](https://github.com/riscv/riscv-gnu-toolchain/files/3390180/err.log)

Let me know if anyone has any idea regarding this or i am missing something?

Th

jim-wilson, 2019-07-14:
If you are using fd in your arch and d in your abi, then you aren't compiling soft-float.

Anyways, your likely problem is that your code was compiled with -mcmodel=medany, but your math library wasn't.  You can get a math library compiled with medany by using --with-cmodel=medany when configuring the toolchain before you build it.

johnnavaj, 2019-07-14:
Yes after compiling with --with-cmodel=medany on the riscv-gnu-toolchain it doesn't give any error. But in fact i need to know how i can replicate the mfloat-abi argument in normal gcc with riscv-gcc ?

jim-wilson, 2019-07-14:
I assume by "normal gcc" you mean the arm gcc port.  For RISC-V, If f or d is in the arch option, then the target has hardware f or d support.  If f or d is in the abi option, then the abi will use the hardware FP registers.  So you can get soft float code by using something like -march=rv64imac -mabi=lp64 which says no hardware FP support and the ABI does not use the FP registers.  You can get softfp code by using something like -march=rv64imafdc -mabi=lp64 which has FP hardware exists but we don't use FP registers for argument passing.  And you get hard float code by using something like -march=rv64imafdc -mabi=lp64d which says FP hardware exists and FP regs are used for argument passing.

The RISC-V scheme is more flexible than the arm scheme and supports additional options.  For instance, you can say -march=rv64imafdc -mabi=lp64f which says that we have both float and double hardware support, but only floats are passed in FP registers.

johnnavaj, 2019-07-15:
Thanks Jim , I need to do some testing from the hardware side as they don't have floating point yet.
So closing this as i have a clear idea on how to proceed.

closed 2019-07-15 by johnnavaj