riscv-collab/riscv-gnu-toolchain #470: getting --> undefined reference to `getrusage'

suhasskrishanmurthy, 2019-07-15:
trying to compile a benchmark code for riscv but getting the follwing error 
linpack.c:(.text+0x2912): undefined reference to `getrusage'
collect2: error: ld returned 1 exit status
but riscv-toolchain has 
riscv_toolchain/riscv-tools/riscv-gnu-toolchain/riscv-glibc/resource/getrusage.c

what am i doing wrong

aswaterman, 2019-07-15:
Are you using the riscv64-unknown-elf compiler or the riscv64-unknown-linux-gnu one?  Only the latter compiler uses glibc.  The former isn't capable of running most POSIX programs.

suhasskrishanmurthy, 2019-07-15:
was using riscv64-unknown-elf compiler. So i changed the compiler and got the executable.Now can i run executable on spike.
because its giving error like 

z  0000000000000000 ra 0000000000010694 sp 000000007f7e9b50 gp 0000000000015850
tp 0000000000000000 t0 0000000000000000 t1 0000000000000030 t2 0000000000015600
s0 0000000000000000 s1 0000000000000000 a0 000000000001073a a1 0000000000000001
a2 000000007f7e9b58 a3 0000000000012ff8 a4 0000000000013050 a5 0000000000000000
a6 000000007f7e9b50 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 ffffffffffffffff t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc fffffffffffffffe va fffffffffffffffe insn       ffffffff sr 8000000200046020
User fetch segfault @ 0xfffffffffffffffe
so this is my code problem or there is still something i am missing

aswaterman, 2019-07-15:
Try using the -static flag on GCC. (The linux compiler defaults to dynamic linking, which doesn’t work on Spike.)

suhasskrishanmurthy, 2019-07-15:
thanks that worked.

billasurendra, 2019-10-01:
I am also facing same issue, I am trying to cross compiling linpackc program with **riscv64-unknown-elf-gcc** compiler, I am getting **undefined reference to `getrusage'** error. When I am cross compiling with **riscv64-unknown-linux-gnu-gcc** i am not getting any error.  Can any please give the technical reason.

Thanks 
Billa Surendra

jim-wilson, 2019-10-01:
getrusage is a system call.  if you are on a linux system, then the linux kernel implements it, and glibc provides a stub that calls into the linux kernel.

If you are on an embedded elf system, you don't have an OS unless you provide one, and hence you have no getrusage support unless you provide it yourself.  Typically you would link with a BSP (board support package) that can implement some system calls by direct accesses to peripherals on the board.  But getrusage is a pretty heavy weight system call, and one that a BSP is unlikely to ever implement.  if you look at the benchmark sources, it probably has support for using other system calls when getrusage isn't available, and these other system calls are likely easier to implement.

billasurendra, 2019-10-03:
Thanks Jim for your  precise explanation.

closed 2021-03-03 by lazyparser