riscv-collab/riscv-gnu-toolchain #471: Softfloat routines and -march

Zeldax64, 2019-07-15:
Hi,

I've noticed that softfloat routines are being generated wrong when using -march=rv32i and -march=rv32im since gcc 8.20.
 When using gcc 8.20, all float routines are generated without mul/div instructions even if using -march=rv32im. 
I downloaded the github toolchain yesterday (gcc version 8.30) and the problem from 8.20 was fixed and now the compiler is generating mul/div instructions in softfloat routines. The problem is that the compiler is still generating mul/div instructions even if -march=rv32i is used.

I've used this simple program to check softfloat routines:
```C
void main() {
	float a, b, c;	
	a = 5.0;
	b = 6.0;
	c = a * b;
}
```

Command line:
```
$ riscv32-unknown-elf-gcc -o mainrv32i.elf main.c -march=rv32i -mabi=ilp32
$ riscv32-unknown-elf-objdump mainrv32i.elf -D > mainrv32i.dump

$ riscv32-unknown-elf-gcc -o mainrv32im.elf main.c -march=rv32im -mabi=ilp32
$ riscv32-unknown-elf-objdump mainrv32im.elf -D > mainrv32im.dump
```

jim-wilson, 2019-07-15:
The -march option affects the code generated by the compiler, but does not affect the C libraries, as those were built at the same time that the compiler was built.  So if you configure your toolchain using --with-arch=rv32i then the float routines will not have mul/div instructions regardless of the compile time -march option.  If you configure your toolchain using --with-arch=rv32im then the float routines will have mul/div instructions regardless of the compile time -march option.

There is another choice here which is to use the configure option --enable-multilib, which will build multiple copies of the libraries for different march options, and then try to link with an appropriate precompiled copy of the libraries.  We can't build every possible arch/abi choice as there are potentially an infinite number of them, but we build about 18 by default, and this includes rv32i/ilp32 and rv32im/ilp32.  So you would get the right result if you configured and built a compiler using --enable-multilib.

Zeldax64, 2019-07-15:
Thanks. I will try this out. I think installation instructions are sometimes not clear since --enable-multilib only appears in linux section and it can also be applied to Newlib.

Zeldax64, 2019-07-31:
@jim-wilson Sorry for the late repply. I want my gcc to support archs rv32ia and rv32ima (other combinations are also welcome). I have tried different combinations of --with-arch and using --enable-multilib but I had no success.

jim-wilson, 2019-08-01:
There is a default list of multilibs that are enabled by --enable-multilib.  This does not include rv32ia and rv32ima.  You could modify this list to include the arches you want.  See the gcc/config/riscv/t-elf-multilib file and the multilib-generator command at the top.

Or you can build two compilers, one configured for rv32ia and one configured for rv32ima.

Zeldax64, 2019-08-12:
Recompiling after changing t-elf-multilib worked as intended. Thanks :D

closed 2019-08-12 by Zeldax64

sudashannon, 2021-09-16:
> There is a default list of multilibs that are enabled by --enable-multilib. This does not include rv32ia and rv32ima. You could modify this list to include the arches you want. See the [...]

Hello jimw:
	I have seen some comments you provided on the github about use multi lib for riscv-gnu-toolchain to solve the problems happen when I use GDB to debug with qemu's s54 machine : bfd requires flen 8, but target has flen 0
	So I follow your suggestion and try to build a 64 bit program for u54 without hard float point support use the command like:
`/home/administrator/bin/riscv_64bit_toolchain/bin/riscv64-unknown-elf-gcc -c -g -Wall -Werror -O0 -Iinc -Iinc/metal/machine/u54 -Iinc/freertos -Isrc/freertos/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -Isrc/freertos/portable/GCC/RISC-V -DMTIME_RATE_HZ_DEF=20000000 _-mabi=lp64_ -mcmodel=medany -ffunction-sections -fdata-sections -fno-common -o build/.objs/xdemo_riscv64/cross/none/debug/src/freertos/portable/Common/mpu_wrappers.c.o src/freertos/portable/Common/mpu_wrappers.c`
	The compile process is successful but the link process is failed as:
`/home/administrator/bin/riscv_64bit_toolchain/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /home/administrator/bin/riscv_64bit_toolchain/bin/../lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/lib/_rv64imafdc/lp64d_/libc_nano.a(lib_a-mlock.o)`
	It seems use the static lib with hard float configuration which makes me confused, could you provide some advise for me plz?
																															Thank u @jim-wilson

jim-wilson, 2021-09-16:
"riscv64-unknown-elf-gcc --print-multi-lib" will show you what multilibs exist in your toolchain.  if the -march and -mabi options you are using are not in this list, then linking with them won't work.  Some libraries like libc and libgcc are precompiled when the toolchain is built, and -march and -mabi on the command line only work if they were precompiled for that -march and -mabi.

You can use --with-arch= and --with-abi= when configuring and building a toolchain to specify the march/mabi you want to use.  This is the easiest solution.

Or you can use --enable-multilib when configuring and building which will build a default set of multilibs which may or may not include the march/mabi you want to use.  There are far too many choices that we can't build them all, we just build a reasonable subset.

Or you can download a toolchain binary that was configured with multilib support.  There is the SiFive toolchain and the GNU MCU Eclipse toolchain for instance.  Both can be found on the sifive.com/software web page.

I also see you are using -mcmodel=medany.  Your precompiled libraries might be mcmodel=medlow.  This is another thing you can specify when configuring and building a toolchain.  The SiFive toolchains are configured mcmodel=medany.  I don't know about the GNU MCU Eclipse toolchains offhand.