riscv-collab/riscv-gnu-toolchain #475: istringstream in C++ stdlib always gives negative numbers

zhemao, 2019-07-19:
The following code will results in incorrect behavior

```
#include <sstream>
#include <stdio.h>

int main(void)
{
    std::istringstream foostream("500");
    int foo;

    foostream >> foo;
    printf("foo = %d\n", foo);

    return 0;
}
```

On x86, this correctly prints

```
foo = 500
```

But running this on riscv-linux produces 

```
foo = -500
```

I've tried this on both Spike and QEMU and they both have the same behavior. 

I'm using commit bce7b5e of riscv-tools.

jim-wilson, 2019-07-19:
>I'm using commit bce7b5e of riscv-tools.

There is your problem.  riscv-tools has an old obsolete and known broken version of the compiler, which should never be used.  This is why the compiler was removed from riscv-tools, because I got tired of telling people to stop using the broken compiler in there, and lobbied for it to be removed.

If you check out and build riscv-gnu-toolchain you will get a working compiler.

aswaterman, 2019-07-19:
Confirmed that this example works with more recent compilers from riscv-gnu-toolchain.

closed 2019-07-19 by aswaterman