riscv-collab/riscv-gnu-toolchain #480: “git submodule update --init --recursive” failed

shjdzc12, 2019-07-23:
I download the sources according to the "Getting the sources" section in riscv/riscv-gnu-toolchain.
I use the commands as follows:

$ git clone https://github.com/riscv/riscv-gnu-toolchain
$ cd riscv-gnu-toolchain
$ git submodule update --init --recursive

When I execute the command "git submodule update --init --recursive", I get following errors：

Cloning into 'boringssl'...
fatal: unable to access 'https://boringssl.googlesource.com/boringssl/': Failed to connect to boringssl.googlesource.com port 443: Connection refused
Clone of 'https://boringssl.googlesource.com/boringssl' into submodule path 'boringssl' failed
Failed to recurse into submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl'
Failed to recurse into submodule path 'qemu/roms/edk2'
Failed to recurse into submodule path 'qemu'

How can I fix this problem?

jim-wilson, 2019-07-23:
The upstream qemu sources use a lot of different git repos as submodules.  This is likely a temporary problem with one of the submodule git repos.  When this happens, just wait a bit, and then try the git submodule command again.  You can try visiting the URL in your browser to see if it is working yet.  I just tried a fresh clone and submodule update, and it worked fine for me.

luoxinjie1146, 2019-09-04:
I had the same problem. I tried to visiting the URL in firfox but it doesn't work.

jim-wilson, 2019-09-04:
If the URL doesn't work, then wait for a while and then try again.  Repeat until it is working.  These problems are usually temporary, and sometimes it may take a few minutes to a few hours for the people on the other side to fix it.  It is working for me at the moment.

If the problem isn't temporary, then you might have a firewall that is blocking access.  You can check whether other access methods like http, ssh, or git work instead.  If that works, you can use an InsteadOf line in your gitconfig to work around the problem.  Worst case you might have to use a VPN or tunnel or something.  But most likely this is a remporary problem.

zarubaf, 2019-10-09:
I am experiencing the same problem. This is a rather annoying issue as it prevents building the toolchain automatically (e.g. when setting up a container). Why is `qemu` needed for building the toolchain? I am sure I am missing a point here, but the easiest seems to just remove it.

jim-wilson, 2019-10-09:
qemu is needed for testing the toolchain, which is very important to the toolchain developers, but not as  important to the toolchain users.

I was able to clone a copy of riscv-gnu-toolchain right now, so it must have been yet another temporary problem.

GQBBBB, 2019-10-30:
> 

It may be the reason for the network, you can open the `.gitmodules` file in `qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl` and change the url of "boringssl" from `https://boringssl.googlesource.com/boringssl` to `https://github.com/google/boringssl.git`.

Then run `git submodule sync` in `qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl`.

Last run `git submodule update --init --recursive`.

: )

Ume-Habiba, 2019-12-11:
when  i am running the following command 
$ git submodule update --init --recursive
I am facing the problems 
remote: Counting objects: 29413, done.
remote: Compressing objects: 100% (10885/10885), done.
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://git.qemu.org/git/capstone.git' into submodule path 'capstone' failed
Failed to recurse into submodule path 'qemu'
computer@computer-virtual-machine:~/riscv-gnu-toolchain$ git submodule update --init --recursive
Cloning into 'capstone'...
fatal: unable to access 'https://git.qemu.org/git/capstone.git/': Could not resolve host: git.qemu.org
fatal: clone of 'https://git.qemu.org/git/capstone.git' into submodule path 'capstone' failed
Failed to recurse into submodule path 'qemu'

please help me to fix them

GQBBBB, 2019-12-14:
Hi Ume-Habiba！

You can try to find the folder where the submodule `capstone` is located, open the `.gitmodules` file in the folder, and then change its url from `https://git.qemu.org/git/capstone.git/` to `https://github.com/aquynh/capstone.git`.

The remaining steps refer to my answer above, hope this is useful for you.

jim-wilson, 2019-12-17:
These kinds of failures tend to be temporary.  Sometimes a network service goes down, and comes back a bit later.  So I'd suggest to just wait a bit and try again when you see a failure like this.

closed 2021-03-03 by lazyparser

AnimeshArya24, 2024-11-23:
git submodule update --init --recursive
Command try to clone git repo https://github.com/Zeex/subhook.git/ but failed .
complete error: 
Username for 'https://github.com': AnimeshArya24
Password for 'https://AnimeshArya24@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/Zeex/subhook.git/' not found
fatal: clone of 'https://github.com/Zeex/subhook.git' into submodule path '/home/kpit/Desktop/Simulation/Qemu/Qemu_build/qemu/roms/edk2/UnitTestFrameworkPkg/Library/SubhookLib/subhook' failed
Failed to clone 'UnitTestFrameworkPkg/Library/SubhookLib/subhook'. Retry scheduled
Cloning into '/home/kpit/Desktop/Simulation/Qemu/Qemu_build/qemu/roms/edk2/UnitTestFrameworkPkg/Library/SubhookLib/subhook'...
Username for 'https://github.com': AnimeshArya24
Password for 'https://AnimeshArya24@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/Zeex/subhook.git/' not found
fatal: clone of 'https://github.com/Zeex/subhook.git' into submodule path '/home/kpit/Desktop/Simulation/Qemu/Qemu_build/qemu/roms/edk2/UnitTestFrameworkPkg/Library/SubhookLib/subhook' failed
Failed to clone 'UnitTestFrameworkPkg/Library/SubhookLib/subhook' a second time, aborting
Submodule path 'roms/ipxe': checked out '4bd064de239dab2426b31c9789a1f4d78087dc63'
Submodule path 'roms/openbios': checked out 'af97fd7af5e7c18f591a7b987291d3db4ffb28b5'
Submodule path 'roms/opensbi': checked out 'a2b255b88918715173942f2c5e1f97ac9e90c877'
Submodule path 'roms/qboot': checked out '8ca302e86d685fa05b16e2b208888243da319941'
Submodule path 'roms/qemu-palcode': checked out '99d9b4dcf27d7fbcbadab71bdc88ef6531baf6bf'
Submodule path 'roms/seabios': checked out 'a6ed6b701f0a57db0569ab98b0661c12a6ec3ff8'
Submodule path 'roms/seabios-hppa': checked out '03774edaad3bfae090ac96ca5450353c641637d1'
Submodule path 'roms/skiboot': checked out '24a7eb35966d93455520bc2debdd7954314b638b'
Submodule path 'roms/u-boot': checked out '840658b093976390e9537724f802281c9c8439f5'
Submodule path 'roms/u-boot-sam460ex': checked out '60b3916f33e617a815973c5a6df77055b2e3a588'
Submodule path 'roms/vbootrom': checked out '0c37a43527f0ee2b9584e7fb2fdc805e902635ac'
Submodule path 'tests/lcitool/libvirt-ci': checked out '77c800186f34b21be7660750577cc5582a914deb'
Failed to recurse into submodule path 'roms/edk2'
Please help to find alternative for this.

I have faced ERROR: Subproject exists but has no meson.build file. error for which i run git recursive command.

TommyMurphyTM1234, 2024-11-23:
> git submodule update --init --recursive Command try to clone git repo https://github.com/Zeex/subhook.git/ but failed . 

Please clarify all of the commands that you use to reproduce this issue. In particular the commands that precede the `git submodule update --init --recursive` command. As far as I know there are no known general issues with cloning and building from this repo and the CI builds are proceeding without any issues. So if you're having a problem then quite likely this is a local issue at your end or with your internet connection.

jordancarlin, 2024-11-23:
It seems like this related to https://gitlab.com/qemu-project/qemu/-/issues/2660. One of the submodules used by QEMU was removed. A new mirror was set up, but the old clone link no longer works. This submodule is not needed for a default build, so the issue is only be encountered when the `--recurse-submodules` flag is used (which shouldn’t be necessary).

This has been patched upstream already, so we should probably switch to newer commit of QEMU to avoid this issue if anyone does ever use `--recurse-submodules`.

TommyMurphyTM1234, 2024-11-25:
Thanks for the explanation @jordancarlin.

The last bump of the QEMU submodule was this:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1549

The lastst release of QEMU is v9.1.2:

- https://www.qemu.org/download/

I tried to bump to this version but don't understand the error that I'm getting:
```
git clone git@github.com:TommyMurphyTM1234/riscv-gnu-toolchain.git bump-qemu -b bump-qemu-from-9.1.0-to-9.1.2
cd bump-qemu
git submodule update --init
cd qemu
git status
HEAD detached at fd1952d814d
nothing to commit, working tree clean
# Bump to v9.1.2: https://gitlab.com/qemu-project/qemu/-/tree/v9.1.2?ref_type=tags
git checkout 508081a49b0d624930ca479b8a27bccdc50bdfb2
fatal: reference is not a tree: 508081a49b0d624930ca479b8a27bccdc50bdfb2
```
I'm not sure how to proceed...

jordancarlin, 2024-12-02:
I was able to successfully bump qemu to 9.1.2. I initially ran into the same issue as you. It seems to be a result of using a shallow clone (which was changed recently I think). I needed to run
```
git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*" 
git remote update
```
to get the new commits to show up.

Unfortunately it seems that the fix for the broken submodule was not back ported to 9.1.2. Not sure if it is worth bumping until a new release is out that actually fixes it. QEMU 9.2 should be out in the next couple of weeks, and I assume will have this fix since it is already merged into master.

TommyMurphyTM1234, 2024-12-05:
> I was able to successfully bump qemu to 9.1.2. I initially ran into the same issue as you. It seems to be a result of using a shallow clone (which was changed recently I think). I needed to run
> ``` [...]

Thanks for that @jordancarlin - much appreciated.

> Unfortunately it seems that the fix for the broken submodule was not back ported to 9.1.2. Not sure if it is worth bumping until a new release is out that actually fixes it. QEMU 9.2 should be out in the next couple of weeks, and I assume will have this fix since it is already merged into master.

Yeah - I think I'll hold off until 9.2 releases.
Shouldn't be long now:

- https://wiki.qemu.org/Planning/9.2