riscv-collab/riscv-gnu-toolchain #482: Separate build process from installation

christian-krieg, 2019-07-26:
It appears that I have to build the toolchain with root privileges because the build process is not separated from the installation process. This poses a serious security concern.

Have you considered to introduce a separate make target 'install' which copies the binaries to INSTALL_DIR? This way, only the `install' target has to be run with root privileges.

Thanks,
Christian

aswaterman, 2019-07-26:
When you configure the toolchain, you can use —prefix to change the install directory to a location you have permission to write.

Because of the nature of this build process (it builds several packages that depend on each other), it would be challenging to do what you suggested.

jim-wilson, 2019-07-26:
The compiler uses relative paths from the install dir.  So you can build with a --prefix pointing at a temporary directory, copy into the final dir, and it will still work.

closed 2021-03-07 by lazyparser