riscv-collab/riscv-gnu-toolchain #483: riscv64-unknown-elf-gcc: error trying to exec 'cc1': execvp: No such file or directory

abhinavfury13, 2019-07-30:
When compiling c file using riscv64-unknown-elf-gcc I am getting this error.

jim-wilson, 2019-07-30:
Did you build the toolchain yourself or did you download one?  Try adding the -v option when running gcc.  This will show what cc1 program it is trying to run.  Does that program exist?

lazyparser, 2021-03-04:
@abhinavfury13 Hi, have you solved this issue?

closed 2021-03-07 by lazyparser

OpenEDF, 2021-09-27:
I also encountered this problem, luckily I found a solution，Because after I downloaded the riscv64-unknow-elf.tar file, I decompressed it into Windows 10 and then transferred it to Ubuntu 20.04 via FTP, so the permissions of all files changed. And when I send the riscv64-unknow-elf.tar file directly to Ubuntu 20.04 without decompression, and then execute decompression in Ubuntu20.04, this can solve this problem.

asdvfsdf, 2023-09-02:
Is this magic?

TommyMurphyTM1234, 2023-09-02:
> Is this magic?

Is what magic?

asdvfsdf, 2023-09-03:
I'm sorry, I was just joking because I think this solution is funny

TommyMurphyTM1234, 2023-09-03:
> after I downloaded the riscv64-unknow-elf.tar file, I decompressed it into Windows 10 and then transferred it to Ubuntu 20.04 via FTP, so the permissions of all files changed. 

Presumably you decompressed onto a Windows NTFS filesystem in which case it's no surprise that Linux file attributes were lost?