riscv-collab/riscv-gnu-toolchain #484: Extension parsing order: "x, s, sx" vs "s, sx, x"

mablinov, 2019-07-30:
Hi all,

I notice in the `riscv_subset_list::parse` function in `riscv-common.c`, line 678, the order of parsing goes

```
  p = subset_list->parse_sv_or_non_std_ext (p, "x", "non-standard extension");
...      
  p = subset_list->parse_sv_or_non_std_ext (p, "s", "supervisor extension");
...   
  p = subset_list->parse_sv_or_non_std_ext
    (p, "sx", "non-standard supervisor extension");
```

and correspondingly, running
`riscv32-unknown-elf-gcc -O2 -march=rv32im_s_sx_xargle -mabi=ilp32 -o /dev/null test.c`
generates
`Fatal error: -march=rv32im_s_sx_xargle: unexpected ISA string at end: xargle`

Surely we want to parse 'x' non-standard extensions at the very end of the string? The latest document, section 27.10: "Non-standard extensions must be listed after all standard extensions."

From my understanding, the parsing order should be "s, sx, (z), x".

kito-cheng, 2019-07-30:
The order was `x`, `s`, `sx` in spec 2.2, and it changed later, it's an incompatible part between 2.2 and latest spec, but I think there is no `s` extension yet? so it would be fine to adjust order.

One solution is add an `-misa-spec=[2.0|2.1|2.2|2.3]` to control that, but the cons is it will make the binutils getting a little messy.

@jim-wilson @asb any comment?

kito-cheng, 2019-07-30:
BTW, clang/llvm is implemented in this order: `x`, `s`, `sx`

mablinov, 2019-08-02:
@kito-cheng what version of the ISA spec do the current GCC/LLVM parsing rules implement?

kito-cheng, 2019-08-02:
Both are based on `2.2`.

And I've start a thread at riscv-sw-dev mailing list for discuss that:
https://groups.google.com/a/groups.riscv.org/forum/?nomobile=true#!topic/sw-dev/aZhMG7NIVTk

jim-wilson, 2019-08-07:
Andrew Waterman tells me that there are no s extensions currently defined, and there are no known sx extensions.  Therefore, he doesn't expect any problems with just changing the parsing to use the new scheme.  Also, sx was dropped in the new spec, so it should just be s, z, x now.

It would still be nice to have -misa-spec for other reasons, e.g. for handling the version dependent csr registers.

lazyparser, 2021-03-04:
@mablinov Hi, based the comments above, have your questions got answered? (If so, I'd close this issue)

closed 2021-03-07 by lazyparser