riscv-collab/riscv-gnu-toolchain #485: About '-march=rv64i'

TresLu, 2019-07-31:
I have compiled successfully:
`./configure --with-arch=rv64i --prefix=/opt/riscv`

and I want to compile 'riscv-test/benchmark/dhrystone/':
`riscv64-unknown-elf-gcc -march=rv64i -I ./../../env/ -I ./../common/ -mcmodel=medany -O2 -fno-common -fno-builtin-printf -o dhrystone.riscv dhrystone.c dhrystone_main.c ./../common/syscalls.c ./../common/crt.S -nostdlib -T ./../common/test.ld -w`

but report:
/opt/riscv64i/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccNVE5LS.o: in function `.L23':
dhrystone_main.c:(.text.startup+0x31c): undefined reference to `__muldi3'
/opt/riscv64i/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccNVE5LS.o: in function `.L0 ':
dhrystone_main.c:(.text.startup+0x334): undefined reference to `__divdi3'
/opt/riscv64i/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: dhrystone_main.c:(.text.startup+0x6f0): undefined reference to `__divdi3'
/opt/riscv64i/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: dhrystone_main.c:(.text.startup+0x724): undefined reference to `__divdi3'
/opt/riscv64i/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccZeQDUq.o: in function `.L47':
syscalls.c:(.text+0x130): undefined reference to `__umoddi3'
/opt/riscv64i/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccZeQDUq.o: in function `.L56':
syscalls.c:(.text+0x160): undefined reference to `__udivdi3'
/opt/riscv64i/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: syscalls.c:(.text+0x170): undefined reference to `__umoddi3'
collect2: error: ld returned 1 exit status

What should I do?

kito-cheng, 2019-07-31:
It seems like something wrong maybe you didn't install correctly or invoke wrong gcc, execute `riscv64-unknown-elf-gcc -v` and it will print out the configuration info for the gcc, it can help to dig out the problem.

jim-wilson, 2019-07-31:
You compiled with -nostdlib which disables all standard libraries.  The missing routines are in libgcc.a which you will have to manually add back in to the link command.  Or compile for a target with the M extension which eliminates the need for libgcc.a.

TresLu, 2019-08-01:
@jim-wilson 
Sorry, would you mind to introduce it detaile? How manually asdd back in to the link command? 

I compiled `./configure --with-arch=rv64im --prefix=/opt/riscv64im`,  and run `riscv64-unknown-elf-gcc -march=rv64im dhrystone` successful, but run `riscv64-unknown-elf-gcc -march=rv64i dhrystone` failed and report the same problem.

jim-wilson, 2019-08-01:
rv64im has a multiply instruction.  rv64i does not.  So rv64i code will call a helper routine to do a multiply.  That helper routine is in libgcc.a.  Which means you have to link with libgcc.a.  GCC links with this by default.  You added -nostdlib which disables that. So you need to manually add libgcc.a back to the compile command.

kito-cheng, 2020-10-12:
Close this issue, it's multi-lib issue, and I think Jim already gave the solution, feel free to create issue or reopen this issue if you have further question.

closed 2020-10-12 by kito-cheng