riscv-collab/riscv-gnu-toolchain #486: Assembler messages: Fatal error: invalid -march= option: `rv64imafdc'

wuliqunyy, 2019-07-31:
Hello,
While I am making /configure --prefix=/opt/riscv
I got this:
Fatal error: invalid -march= option: `rv64imafdc'

What I did wrong?
Thx,
Liqun

jim-wilson, 2019-07-31:
There isn't enough info here to tell what went wrong.  I need the exact error message and the command from the build log that produced it.  I might also need some info about the context that command was run in.  If you can provide an entire build log that should have enough info.

wuliqunyy, 2019-08-01:
Hi Jim,
Indeed, I have to mention that I am trying to build with MSYS2 + MINGW. I am not sure I have all dependencies right, though. 

here is the log:
[log.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3455581/log.txt)

Thanks,
Liqun

jim-wilson, 2019-08-01:
The error message is actually
    Assembler messages:
    Fatal error: invalid -march= option: `rv64imafdc'
That "assembler messages" is the key, because this is an assembler error not a compiler error.  This means that the riscv gcc you built can't find the riscv as.  You are accidentally running an x86 as instead, which obviously doesn't support riscv -march options.

The way this should work is that there should be a $prefix/riscv64-uknown-elf/bin/as binary which is the riscv assembler that will be used by the compiler.  You can see the assembler used by the compiler by running gcc with -v.  So if I do
    touch tmp.c
    riscv64-unknown-elf-gcc -v tmp.c
then in the output I see
    /home/jimw/FOSS/install-riscv64/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/as -v --traditional-format -march=rv64gc -mabi=lp64d -o /tmp/ccI1oWNJ.o /tmp/ccwntNGw.s
where /home/jimw/FOSS/install-riscv64 is my prefix.  The compiler uses relative paths so that you can move it and it will still work.  You need to look at your install ltree and figure out why this isn't working for you.

Don't add $prefix/riscv64-unknown-elf/bin to your path to try to fix this, as this will break the native x86 gcc compiler.

I don't have access to a windows machine at work, so I haven't done any testing of builds on windows.  We build our windows binaries by cross compiling from an ubuntu linux machine, which is an easier way to build them.

jim-wilson, 2019-08-01:
Running gcc with -print-search-dirs will print the list of dirs that gcc searches for programs and libraries.  One of the dirs on the programs line should be the dir that the as is installed into.  If there is a problem here, you could try copying as into one of the other dirs, e.g. the dir that contains the cc1 binary.

wuliqunyy, 2019-08-05:
Hi Jim,
Thanks for your help.
I fix the path issue by using the absolute PATH for riscv bin. The Assembler issue seems to be gone.
But now I run into gdb issue again.
Is this also the path issue?
[log.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3466529/log.txt)

jim-wilson, 2019-08-05:
The gdb tui feature (terminal-window user interface) requires a curses package.  The curses packages are usually not installed by default.  You have to install one, probably ncurses, ot disable the tui feature via the configure option --disable-tui.  Though tui should be disabled by default if the curses library is missing, so not clear what is wrong.  But manually fixing this by installing the  missing ncurses package ot disabling tui should fix the build.

wuliqunyy, 2019-08-07:
Thank you Jim.
I solved the issue in mingw:
pacman -S mingw-w64-x86_64-pdcurses
Now I have everything built successfully!

closed 2019-08-07 by wuliqunyy

YoussefGuirat, 2020-12-28:
Hi,
I have a similar issue here .
I'm using STM32mp157c-dk2 discovery board and I'm following [Bootlin lab](https://bootlin.com/doc/training/embedded-linux-4d/embedded-linux-4d-labs.pdf) . Now I'm trying to make the uboot which I downloaded from here : https://gitlab.denx.de/u-boot/u-boot.
When I run make under u-boot directory I got this log: 
[image]

jim-wilson, 2020-12-30:
This is a support forum for RISC-V toolchains, not ARM toolchains.