riscv-collab/riscv-gnu-toolchain #487: How to get Each Risc-v instruction's machine cycle & respective size of Instruction's

cghalibkhan, 2019-07-31:
Hi Every one,
                 * Actually, i want to know Risc-v Instructions machine cycle and their respective size.

jim-wilson, 2019-07-31:
Cycles per instruction depends on the target hardware.  You would have to check the hardware documentation for the target.

It isn't clear what you mean by instruction size.  Instructions are 32-bit in the base architecture, and some instructions have a 16-bit encoding in the C (compressed) extension.  If you use objdump -dr to disassemble, it will show you whether instructions are two or 4 bytes.  Instruction data can be 32-bit for the rv32i base architecture and 64-bit for the rv64i base architecture.

If you want machine cycles, you need to use a cycle accurate simulator.  Or alternatively, you can try reading the cycle (and for rv32 cycleh) registers.

cghalibkhan, 2019-08-02:
Hi good morning,
     
1> I check hardware Documents, in that there is no cycle info in it.

2> I checked cycle-accurate simulator, but it's showing a paid version of simulators, & am not sure that those simulators are for Risc-v arch compactable.

Help me to sort this issue.

Thanks,
khan.

jim-wilson, 2019-08-02:
Different implementations of the RISC-V architecture have different cycle timing.   I can't help you with that.  You must choose an implementation first, and then look for info on that implementation.  You haven't clearly specified an implementation.

If you can get access to rtl or verilog, then you can do cycle accurate simulations.  This type of simulation will be slow.  You have to have detailed hardware info to get a cycle accurnate simulator, which means it has to be a hardware vendor simulator, or a rtl/verilog style simulator.  Otherwise, you can't get cycle accurate simulation results.

As a compiler guy, I can only offer vague approximations, e.g. most alu instructions will be single cycle, multiply will be a few to several cycles, divide will be a few to several times slower than multiply, load/stores will be 2 cycles or slower, and FP instructions if present will probably be a few times slower than the corresponding integer instructions.  But this stuff will all depend on the target, and can vary greatly from one target to the next.

You did mention the SiFive U540, and the SiFive u540 docs do have some hardware timing info in them.  But this isn't a forum for SiFive u540 discussions.

I already pointed you at the cycle counter csr which is a way to get info from the hardware if you have hardware.

lazyparser, 2021-03-04:
@cghalibkhan Hello, have you got the result you want?

cghalibkhan, 2021-03-04:
No.

[EXT]

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year and it seems that @jim-wilson already adequately addressed the issue. No RISC-V GCC toolchain issue identified.

closed 2022-09-27 by TommyMurphyTM1234