riscv-collab/riscv-gnu-toolchain #488: How to change the arch and abi combinations when using the option "--enable-multilib" ?

hyf6661669, 2019-08-01:
At present, using "--enable-multilib" will generate several combinations of different ```arch``` and ```abi```, some of them are not useful for me and some combinations I need are not included. Could you please tell me how to edit these combinations?

Thanks!

jim-wilson, 2019-08-01:
I'm assuming that you are building an embedded elf toolchain.

If you look at riscv-gcc/gcc/config/riscv/t-elf-multilib you will find the definitions used to generate the default multilib list.  The conventions here are a little complicated, but we have a python script called multilib-generator to generate the file, and the command used to generate the default file is given at the top.  So you can take the multilib-generator command from the t-elf-multilib file, edit it, use it to generate a new t-elf-multilib file, and rebuild gcc.  You can check the end result with "riscv64-unknown-elf-gcc --print-multi-lib".

palmer-dabbelt, 2019-08-01:
I wrote a blog on this: https://www.sifive.com/blog/all-aboard-part-5-risc-v-multilib

closed 2019-08-04 by hyf6661669