riscv-collab/riscv-gnu-toolchain #495: Is there any avaliable efficient softfloat library for risc-v ?

hyf6661669, 2019-08-23:
When I'm using "float" variable in my C codes,  I find that the generated codes(using -march=rv32imc, -mabi=ilp32) of riscv-compiler are much larger than the codes generated by arm-compiler(the target core is cortex-m3). And the execution time is also longer.

jim-wilson, 2019-08-23:
The arm soft-float support is hard written assembly code that implements a subset of IEEE FP.  The RISC-V soft-float support is using a C implementation that implements the full IEEE FP spec.  So yes, the RISC-V one is larger and slower.

There is also another issue that arm EABI has 64-bit long double and RISC-V has 128-bit long double, which means arm has 2 soft-float libraries and RISC-V has 3 soft-float libraries.  That can be an issue if you are calling printf, and printf then pulls in all of the soft-float code.  The plan is to fix this with a new embedded ABI that has fewer arg registers for lower latency interrupt handling, and reduces long-double to 64-bits to reduce the soft-float code bloat.

If you want a different soft-float library, you might try asking elsewhere, like the sw-dev mailing list.  The only RISC-V soft-float support in riscv-gnu-toolchain is the one that the riscv-gnu-toolchain is already using.

hyf6661669, 2019-08-23:
@jim-wilson 
Thanks.

closed 2019-08-23 by hyf6661669