riscv-collab/riscv-gnu-toolchain #496: How to install the riscv-gcc tool chain

munish37, 2019-08-23:
Hi
I am trying to install riscv-gcc tool chain by following below commands
1.git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
2.sudo yum install autoconf automake libmpc-devel mpfr-devel gmp-devel gawk bison flex texinfo patchutils gcc gcc-c++ zlib-devel expat-devel
3../configure --prefix=/opt/riscv
4. make linux

While following these above steps and after running step 4.i am getting compilation error
cc1: internal compiler error: Aborted.
Can you please tell correct steps for installing riscv-gcc toolchain
Can you also please tell after above toolchain installatiom. what should be the value of env variable RISCV_GCC.

Regards
Munish

hyf6661669, 2019-08-23:
1.You'd better display all the warning/error information

2. Some Makefiles would use ```RISCV_GCC``` to set the path of the compiler, maybe you should look at Makefile for details.

jim-wilson, 2019-08-23:
It is also helpful to know what system you are building it on, and what commit id you have.  I did a build yesterday on an x86 Ubuntu 16.04 system and it worked for me.

munish37, 2019-08-26:
Steps followed to reproduce the error in last
1.git clone https://github.com/google/riscv-dv
2. In the riscv-dv directory
   git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
   New folder riscv-gnu-toolchain gets created inside riscv-dv
3.yum install autoconf automake libmpc-devel mpfr-devel gmp-devel gawk bison flex texinfo 
4. Since my machine is CentOS 64 bit ran below
   ./configure --prefix=/opt/riscv
5. make linux
while running make linux getting below error
/home/munish/Documents/riscv-dv/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/home/munish/Documents/riscv-dv/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -xc -nostdinc /dev/null -S -o /dev/null -fself-test=../.././riscv-gcc/gcc/testsuite/selftests
pure virtual method called
terminate called without an active exception
cc1: internal compiler error: Aborted
0xbbc8e5 crash_signal
	../.././riscv-gcc/gcc/toplev.c:326.  

Please correct me if i am missing any steps 

Regards
Munish

jim-wilson, 2019-09-05:
You didn't mention the Centos version number.  Assuming you have Centos 6, which uses gcc-4.4 as the default compiler, the problem may be that your gcc version is too old to compile the gcc-9 that is now in riscv-gnu-toolchain.  If you can install newer compiler versions that might solve the problem. 
 Alternatively, you could try checking out older versions of the toolchain.  Unfortunately we don't create useful tags or branches for this, but the last commit before the gcc-9 upgrade is 0914ab9.
That gives you a gcc-8.3 instead of gcc-9.

willhaihui, 2019-10-22:
Reading the thread, I don't know Munish's reported issue has been resolved. I'm running into similar issue on CentOS 6.6 with gcc-9.2.0. The stack trace is shown as below. How can I resolve it?

% make linux
...

/opt/riscv-dv/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/opt/riscv-dv/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -xc -nostdinc /dev/null -S -o /dev/null -fself-test=../.././riscv-gcc/gcc/testsuite/selftests
pure virtual method called
terminate called without an active exception
cc1: internal compiler error: Aborted
0xbbc8e5 crash_signal
	../.././riscv-gcc/gcc/toplev.c:326
0x1345f32 print_any_labels
	../.././riscv-gcc/gcc/diagnostic-show-locus.c:1457
0x1345f32 print_line
	../.././riscv-gcc/gcc/diagnostic-show-locus.c:2225
0x1345f32 diagnostic_show_locus(diagnostic_context*, rich_location*, diagnostic_t)
	../.././riscv-gcc/gcc/diagnostic-show-locus.c:2310
0x1349da2 test_one_liner_labels
	../.././riscv-gcc/gcc/diagnostic-show-locus.c:2786
0x134abbf test_diagnostic_show_locus_one_liner
	../.././riscv-gcc/gcc/diagnostic-show-locus.c:2833
0x1359827 selftest::for_each_line_table_case(void (*)(selftest::line_table_case const&))
	../.././riscv-gcc/gcc/input.c:3550
0x1346a12 selftest::diagnostic_show_locus_c_tests()
	../.././riscv-gcc/gcc/diagnostic-show-locus.c:3726
0x12d0ca4 selftest::run_tests()
	../.././riscv-gcc/gcc/selftest-run-tests.c:91
0xbbce82 toplev::run_self_tests()
	../.././riscv-gcc/gcc/toplev.c:2269
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
make[2]: *** [s-selftest-c] Error 1
rm fsf-funding.pod gcov.pod gpl.pod cpp.pod gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod
make[2]: Leaving directory `/opt/riscv-dv/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc'
make[1]: *** [all-gcc] Error 2
make[1]: Leaving directory `/opt/riscv-dv/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [stamps/build-gcc-linux-stage1] Error 2

Thanks
Will

jim-wilson, 2019-10-24:
Centos provides old versions of the tools that probably can't build the toolchain.  However, there is an alternate set of tools available that provide a current GNU toolchain for Centos that should work for the RISC-V toolchain build.  See the instructions at
   https://www.softwarecollections.org/en/scls/rhscl/devtoolset-7/

weedmank, 2020-01-12:
Try this procedure.  http://hdlexpress.com/RisKy1/How2/toolchain/toolchain.html

kito-cheng, 2020-10-12:
Close this issue due to long time no further discussion/response, feel free to open new issue if you got same problem with latest riscv-gnu-toolchain.

closed 2020-10-12 by kito-cheng