riscv-collab/riscv-gnu-toolchain #497: gp relaxation does not work for symbols with alignment > 1024

anatol, 2019-08-23:
Here is a small example that shows the issue we experience.

```c
#include <stdint.h>

#ifdef NOT_VERY_RELAXING
// This one silently makes ld stop relaxing
uint8_t shmem_buf_sram04[2048] __attribute__((aligned(2048), section(".shmem_buffers")));
#else
// These are fine
uint8_t shmem_buf_sram02[1024] __attribute__((aligned(1024), section(".shmem_buffers")));
uint8_t shmem_buf_sram03[1024] __attribute__((aligned(1024), section(".shmem_buffers")));
#endif

__attribute__((section(".sbss"))) uint32_t var_sbss;
__attribute__((section(".sdata"))) uint32_t var_sdata = 1;
uint32_t var_bss[16];
uint32_t var_data[16] = {1};

int main(void)
{
    return var_sbss + var_sdata + var_bss[0] + var_data[0];
}
```

```ld
MEMORY {
  SRAM (!RX): ORIGIN = 0x00010000, LENGTH = 0x20000
}

SECTIONS {
  .sbss : {
    __global_pointer$ = .;
    *(.sbss*)
  } > SRAM

  .sdata : { *(.sdata*) } > SRAM
  .bss : { *(.bss*) } > SRAM
  .data : { *(.data*) } > SRAM
  .text : { *(.text*) } > SRAM

  .shmem_buffers : {
    . = ALIGN(2048);
    *(.shmem*)
  } > SRAM
}
```

```sh
#!/bin/bash

set -e

rm -f main[12]*

riscv32-unknown-elf-gcc -c main.c -o main1.o
riscv32-unknown-elf-ld -T linker.ld main1.o -o main1
riscv32-unknown-elf-objdump -D -S -j.text main1

riscv32-unknown-elf-gcc -c main.c -o main2.o -DNOT_VERY_RELAXING
riscv32-unknown-elf-ld -T linker.ld main2.o -o main2
riscv32-unknown-elf-objdump -D -S -j.text main2
```

In this example we expect that access to symbol `shmem_buf_sram04` is relaxed. But linker for some reason does not do it. See op at offset `10094` below.

It looks like any alignment bigger than 1024 triggers this behavior. And it sounds like a toolchain issue.

```
main1:     file format elf32-littleriscv

Disassembly of section .text:

00010088 <main>:
   10088:	1141                	addi	sp,sp,-16
   1008a:	c622                	sw	s0,12(sp)
   1008c:	0800                	addi	s0,sp,16
   1008e:	0001a703          	lw	a4,0(gp) # 10000 <var_sbss>
   10092:	0041a783          	lw	a5,4(gp) # 10004 <var_sdata>
   10096:	973e                	add	a4,a4,a5
   10098:	00818793          	addi	a5,gp,8 # 10008 <var_bss>
   1009c:	439c                	lw	a5,0(a5)
   1009e:	973e                	add	a4,a4,a5
   100a0:	04818793          	addi	a5,gp,72 # 10048 <var_data>
   100a4:	439c                	lw	a5,0(a5)
   100a6:	97ba                	add	a5,a5,a4
   100a8:	853e                	mv	a0,a5
   100aa:	4432                	lw	s0,12(sp)
   100ac:	0141                	addi	sp,sp,16
   100ae:	8082                	ret

main2:     file format elf32-littleriscv

Disassembly of section .text:

00010088 <main>:
   10088:	1141                	addi	sp,sp,-16
   1008a:	c622                	sw	s0,12(sp)
   1008c:	0800                	addi	s0,sp,16
   1008e:	0001a703          	lw	a4,0(gp) # 10000 <var_sbss>
   10092:	67c1                	lui	a5,0x10
   10094:	0047a783          	lw	a5,4(a5) # 10004 <var_sdata>
   10098:	973e                	add	a4,a4,a5
   1009a:	67c1                	lui	a5,0x10
   1009c:	00878793          	addi	a5,a5,8 # 10008 <var_bss>
   100a0:	439c                	lw	a5,0(a5)
   100a2:	973e                	add	a4,a4,a5
   100a4:	67c1                	lui	a5,0x10
   100a6:	04878793          	addi	a5,a5,72 # 10048 <var_data>
   100aa:	439c                	lw	a5,0(a5)
   100ac:	97ba                	add	a5,a5,a4
   100ae:	853e                	mv	a0,a5
   100b0:	4432                	lw	s0,12(sp)
   100b2:	0141                	addi	sp,sp,16
   100b4:	8082  
```

I tried both 8.3 and 9.2 from the current master branch and the behavior is the same.

aswaterman, 2019-08-23:
The issue is that, during link time, the distance from gp to the symbol might change, causing the relocation to overflow.  One of the things that affects how far the symbol might move from gp is the alignment of the symbol.

Consider the case that gp = 0x1800 and sym = 0x2000, and sym has alignment 0x1000.  During relaxation, we decide to delete 16 bytes located below 0x1800.  gp now decreases to 0x17f0, but sym must stay at 0x2000 because of its alignment constraint.  Now, the distance from gp to sym exceeds the maximum displacement allowed for gp-relative addressing.

We haven't found this to be much of a problem, because most symbols have alignment <= 16 bytes.

rspangit, 2019-08-24:
The symbols which are failing to relax do not specify alignment.  var_sdata and var_sbss are 32-bit symbols with default (32-bit) alignment.  It's the presence of *another* symbol with alignment >=2K anywhere in the linker script which seems to prevent relaxation of variables in any section other than the one where __global_pointer$ is defined.

If you move this line from .sbss to .sdata, then var_sdata will relax, and var_sbss will no longer relax:
`__global_pointer$ = .;`

If instead you move it to a dummy section, then neither var_sdata nor var_sbss will relax when the 2048-aligned shmem buffer is present:
```
.glo : {
    __global_pointer$ = .;
  } > SRAM

  .sbss : { *(.sbss*) } > SRAM
  .sdata : { *(.sdata*) } > SRAM
```

jim-wilson, 2019-09-05:
Normally sbss is near the end of the executable file, which means the alignment of most any section can affect relaxation to gp.  You have an unusual situation where sbss is at the start.  This prevents the linker from reducing executable file size, since normally sbss will have a file size of zero and a non-zero memory size.  Since you have sbss at the start, alignment of sections after it won't affect it.  We don't have code to handle that case.  We would have to add code to check to see where sbss is in the object file, and which sections occur before it.  Currently, we just use the max section alignment for the entire file if gp and the symbol aren't in the same section.

The code in question incidentally can be found in binutils, in bfd/elfnn-riscv.v in the function _bfd_riscv_relax_lui.  There is an optimization where it check to see if gp and the symbol are in the same section in which case it can use that section's alignment.  Otherwise it uses the maximum alignment of any section in the file, which is computed by _bfd_riscv_get_max_alignment and passed in as an argument.  This last part could perhaps instead be the max alignment of any section that occurs before gp or the symbol, which ever one occurs later in the file.

  if (gp)
    {
      /* If gp and the symbol are in the same output section, which is not the  
         abs section, then consider only that output section's alignment.  */
      struct bfd_link_hash_entry *h =
        bfd_link_hash_lookup (link_info->hash, RISCV_GP_SYMBOL, FALSE, FALSE,
                              TRUE);
      if (h->u.def.section->output_section == sym_sec->output_section
          && sym_sec->output_section != bfd_abs_section_ptr)
        max_alignment = (bfd_vma) 1 << sym_sec->output_section->alignment_power;
    }

I have no plans to try to improve this, just pointing out the work that needs to be done.

By the way, you should be using "__global_pointer$ = . +  0x8000;" so that you can use a 4KB range gp +/- offset.  With the current setting, you only get a 2KB range gp+offset.

aswaterman, 2019-09-05:
Edited Jim’s response to amend +0x8000 to +0x800.

lazyparser, 2021-03-04:
@anatol would you like to share the status of the issue? Is it got solved?

TommyMurphyTM1234, 2022-09-29:
No update/activity in over a year (and, really, in over three years) and the reasons for the allegedly incorrect behaviour have been explained clearly by @jim-wilson and @aswaterman.

closed 2022-09-29 by TommyMurphyTM1234