riscv-collab/riscv-gnu-toolchain #498: Getting linker ld Error

cghalibkhan, 2019-08-27:
Hi,

             I building a Riscv Toolchain with upstream's sources, Tool is built successfully but when to execute riscv64-unknown-linux-tool facing an issue with a linker, got an error as below.

Error:
<PATH>/Exe-linux-elf/lib/gcc/riscv64-unknown-linux-gnu/9.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find <PATH>/Exe-linux-elf/sysroot/lib/libc.so.6 inside <PATH>/Exe-linux-elf/sysroot
<PATH>/Exe-linux-elf/lib/gcc/riscv64-unknown-linux-gnu/9.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find <PATH>/Exe-linux-elf/sysroot/usr/lib/libc_nonshared.a inside <PATH>/Exe-linux-elf/sysroot
<PATH>/Exe-linux-elf/lib/gcc/riscv64-unknown-linux-gnu/9.1.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find <PATH>/Exe-linux-elf/sysroot/lib/ld-linux-riscv64-lp64d.so.1 inside <PATH>/Exe-linux-elf/sysroot
collect2: error: ld returned 1 exit status

--------------------------------------------------------------
Build a Tool with the following configuration.
---------------------------------------------------------------

[Build-riscv-tool.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3544447/Build-riscv-tool.txt)

* can anyone please help me to resolve this issue,

Thanks,
Ghalib khan.

lazyparser, 2021-03-04:
@cghalibkhan Hello, have you fixed it? Would you like to share the progress?

cghalibkhan, 2021-03-04:
This issue while raised when you try to build different version combination of gcc, binutiles & glibc. soo try with correct combination of sources.

[EXT]

closed 2021-03-05 by lazyparser