riscv-collab/riscv-gnu-toolchain #499: Build toolchain with glibc 2.30

cghalibkhan, 2019-08-27:
Hello,
We want to build riscv linux toolchain with glibc 2.30. 
Currently, riscv-gnu-toolchain has glibc 2.29.
Is this the right approach to change to glibc 2.30? Please suggest
1. Clone https://github.com/riscv/riscv-gnu-toolchain
2. Overwrite riscv-gnu-toolchain/riscv-glibc with glibc 2.30( http://mirrors.kernel.org/gnu/glibc/)
3. ./configure --prefix=<my path>
4. make linux

Got toolchain installed and it works.

Thanks

cghalibkhan, 2019-08-28:
Thanks.
Can you please elaborate on version dependencies? How can we make sure all the versions we are considering are correct? Also, are there any errors we get during toolchain build or runtime?

cghalibkhan, 2019-08-30:
> Note that since the rv32 support is not upstream yet, a glibc-2.30 can only work for rv64.
> 
> I like to use git remote, so for instance for binutils I do
> cd riscv-binutils
> git remote add upstream git://sourceware.org/git/binutils-gdb.git
> git fetch upstream
> git checkout upstream/master
> which gives me binutils top of tree. I do this for development work. You shouldn't do this for a release of course, you should use a binutils release branch. The same trick can work for glibc. It is just a different URL for upstream and a different branch name.
> 
> When changing versions, you need to be aware of version dependencies. Sometimes a gcc upgrade requires a binutils upgrade, etc. Sometimes a tool upgrade might require top level makefile changes. But this stuff is less common.

is there any difference in the release version & master version, if there what are them.

jim-wilson, 2019-08-30:
Development work is done on master.  For releases, we create branches, test, fix bugs, and release.  The release may be tagged.  If bugs are found after release, then patches are added to the release branch, and new releases as made.

The main difference between master and release branches is that release branches are supported, and master is not.  If you build something using master, find a bug a month later, and report it, then they will tell you to upgrade to current master and try again, because it might already be fixed.  Then they will point you at info on how to do development work for that package.  If you build something using a release branch, find a bug a month later, and report it, then you will get a pointer to the fix if it already exists, or a fix will be written, or maybe you get a workaround.  Anyways, you get a much more useful answer if you are using a release branch.

closed 2021-03-07 by lazyparser