riscv-collab/riscv-gnu-toolchain #500: Is this repo still the correct repo for a cross compiler?

marcopeereboom, 2019-08-29:
My notes no longer seem to produce a cross compiler with the expected tuple riscv64-unknown-elf but produce riscv64-unknown-linux-gnu. Before I am going to spend time figuring this out I'd like to know if this repo still is the official riscv linux cross compiler.

Thanks.

jim-wilson, 2019-08-29:
Personally, I think the official sources are the FSF ones.  But I am still helping to maintain this git tree.

If you use --disable-linux when configuring, you will get a risc-v embedded elf cross compiler when you run "make".  If you use --enable-linux when configuring, you will get a risc-v linux cross compiler when you run "make".  --disable-linux is the default.  All this does is control whether you get a "all: linux" or "all: newlib" line in the Makefile.

If you run 'make newlib" you will get an risc-v embedded elf cross compiler.  If you run "make linux" you will get a risc-v linux cross compiler.  This is true regardless of whether you used a disable or enable linux configure option.

Personally, I prefer to use the configure options, and have separate build trees for embedded elf and linux.  That way I don't have to remember if I last typed "make newlib" or "make linux".  It is always just "make".

marcopeereboom, 2019-08-30:
Thank you this helps a lot.

And just to make sure, I really should switch to upstream FSF source, right?

jim-wilson, 2019-08-30:
Long term, I'd like to eliminate a lot of these riscv/riscv-* trees, but we may keep riscv/riscv-gnu-toolchain as a useful build script, and just have it use FSF trees as submodules.

If you want to build a 32-bit linux cross compiler, you have to use riscv/riscv-glibc as this support isn't upstream yet.  If you want gdb sim, then you have to use riscv/riscv-binutils-gdb as this support isn't upstream yet.  Otherwise, everything else is upstream, and being maintained upstream, except for support for draft extensions like vector and bitmanip, and those could just as well be branches in the FSF trees, we just haven't tried to do that yet.

closed 2019-09-03 by marcopeereboom