riscv-collab/riscv-gnu-toolchain #501: Can't link hard-float modules with soft-float modules

zhuanhao-wu, 2019-09-03:
Hi, I'm trying to build gcc from the master branch with the following two commands since I don't want to generate FP instructions :
`./configure --prefix=/opt/riscv/ --with-arch=rv64ia --with-abi=lp64`
`make linux -j`

However, the compilation fails with the following message:
```
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /opt/riscv-gnu-toolchain/build-glibc-linux-rv64ia-lp64/support/links-dso-program.o: can't link double-float modules with soft-float modules 
/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /opt/riscv-gnu-toolchain/build-glibc-linux-rv64ia-lp64/support/links-dso-program.o collect2: error: ld returned 1 exit status ../Rules:193: recipe for target /opt/riscv-gnu-toolchain/build-glibc-linux-rv64ia-lp64/support/links-dso-program' failed 
```

I wonder was it that I was not specifying a valid arch/abi combination?

jim-wilson, 2019-09-03:
I would expect this to work.  It isn't obvious what is wrong.

I would guess the problem is that you already have a compiler in /opt/riscv, which was configured for hard float, and the build system got confused about whether you were building for hard float or soft float because both compilers existed during the build.  The fix for this is that you should always point --prefix at an empty dir when you do a build, so that there is only ever one compiler during the build.

zhuanhao-wu, 2019-09-03:
Thank you, this solved my problem.

Do you think it's worth mentioning the --prefix in the README page?

zhuanhao-wu, 2019-09-04:
I will close this issue as my problem is solved.

closed 2019-09-04 by zhuanhao-wu

jim-wilson, 2019-09-05:
Yes, some docs on prefix would be useful, this has come up before, which is why I was able to guess the answer.  It is OK to use the same prefix for a newlib and linux compiler.  it isn't Ok to use the same prefix for a rv32 and rv64 linux compiler, because they use the same sysroot dir name and will conflict.  I think it is OK to use the same prefix for a rv32 and rv64 newlib compiler because they use different sysroot dir names.  And as I already mentioned, it isn't OK to use the same prefix if you have two compilers with the same rv32/rv64 and newlib/linux choices, but different default march/mabi choices.  This may fail when building the second compiler.

I haven't gotten around to trying to write some docs yet as I'm extremely busy.

mathmax12, 2019-09-19:
Hi 
My installation goes well for the 
`./configure --prefix=/opt/riscv
make`

But when I try to install the linux compiler,
```
./configure --prefix=/opt/riscv
make linux
```
I got the same error.
``/opt/riscv/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: failed to merge target specific data of file /opt/riscv-gnu-toolchain/build-glibc-linux-rv64ia-lp64/support/links-dso-program.o collect2: error: ld returned 1 exit status ../Rules:193: recipe for target /opt/riscv-gnu-toolchain/build-glibc-linux-rv64ia-lp64/support/links-dso-program' failed `
`
I even tried to changed the --prefix to another path , i.e. --prefix=/opt/riscv2.
But the error persisted.

Any suggestion will be appreciated.

jim-wilson, 2019-09-19:
The error message in your report doesn't match the error message in the title, and this is a closed issue.  New issues for new problems please.

The error message mentions build-glibc-linux-rv64ia-lp64 which is not the default arch and abi, but you didn't mention using arch and abi options.

It also looks like you trimmed the error message.  The very first error message is usually the most important, and one usually can't tell much from the cascading errors that followed, but all you gave above is one of the cascading errors.

Also, the error message refers to a links-dso-program.o file, but I see no such file created during the build.  There is a links-dso-program-c.o file though.  Maybe you are on a branch?  Or maybe you are using an older version of the tree?

In any case, I don't have enough info to reproduce the problem, so I can't say what might be wrong.

mathmax12, 2019-09-20:
Thanks for replying.
I will open a new thread.