riscv-collab/riscv-gnu-toolchain #504: RISCV GNU toolchain regression

meghamegha, 2019-09-06:
Hello,
We built linux riscv-gnu-toolchain and after that started with regression.
-make linux
-make check-gdb-linux

We are finding 'check-gdb-linux' target takes more time to finish .Attached (gdb.log) is the partial log. Is this expected? Could you please help us figure the problem?

For GCC/binutils regression, are there any reference results with which we can compare our results?

Thanks,
megha
[gdb.log](https://github.com/riscv/riscv-gnu-toolchain/files/3582368/gdb.log)

jim-wilson, 2019-09-06:
There is no one maintaining check-gdb-linux, so you can't expect it to work.  There is also no one officially maintaining the gdb linux port or the qemu gdb stub support.  I work on them as time permits, which is usually not much time, as I have far too many other things to do.

I am doing occasional gdb linux testing natively on a HiFive Unleashed.  I did a run a couple of days ago, and got this result which is reasonable.

                === gdb Summary ===

# of expected passes            59253
# of unexpected failures        1161
# of unexpected successes       2
# of expected failures          79
# of unknown successes          5
# of known failures             85
# of untested testcases         93
# of unresolved testcases       10
# of unsupported tests          244

I had the unexpected failures below 500 at one point, but the gdb developers are adding new features and new testcases faster than I can fix them, as I have far too many things to do.

As for the amount of time that check-gdb-linux takes, the more testcases gdb has the longer it takes to run them all.  Gdb developers are pretty good at adding new testcases for new features.  Also, when gdb/qemu didn't work very well, check-gdb-linux was probably fast because lots of stuff failed right away.  Now that it is mostly working, a lot of stuff runs which takes more time.  Also, some stuff runs and then fails, which takes even more time to recognize the failure and handle it, maybe waiting for a time out which takes a long time.  As gdb support improves, we should have fewer failures, and fewer timeouts, and the testsuite may actually run faster even though more stuff is being run.  But meanwhile, until someone is actively maintaining it, check-gdb-linux is probably not very useful.

jim-wilson, 2019-09-06:
As for the binutils and gcc testsuite results, these are maintained, and we have whitelists in the test dir, most of which have comments indicating why each test fails when the failure cause is known.  You can compare test results against the whitelist by using "make report-linux".  If you have already done the tests it should just check the whitelists against your existing results, otherwise it will run the tests from scratch and then compare.

Note that if you cd into the regression dir and type "make" it will build 30 or so toolchains for different arch/abi combinations, run the testsuite on each one, run make report on each one, and then error if there are any unexpected failures that aren't in the white lists.  This is the current gold standard for testing gcc patches.

TommyMurphyTM1234, 2022-09-29:
No update/activity in over three years and @jim-wilson explained the issues the time clearly.

closed 2022-09-29 by TommyMurphyTM1234