riscv-collab/riscv-gnu-toolchain #506: Analyse GCC regression fails

meghamegha, 2019-09-11:
Hello,
We have built RISCV linux GNU toolchain(binutils_version=2.32 gcc_version=9.1.0 glibc_version=2.30) and run regression tests(make report-gcc). There are few fails.
Can someone please help us in analysing them?

Snippet of report-gcc:
---------------------------
=== g++: Unexpected fails for rv64imafdc lp64d medlow ===
FAIL: c-c++-common/Warray-bounds-2.c  -std=gnu++98 memcpy (test for warnings, line 90)
FAIL: c-c++-common/Warray-bounds-2.c  -std=gnu++98  (test for warnings, line 95)
..

This is our try:
On console executed the following:
 /var/lib/jenkins/workspace/riscv64-gnu-toolchain/Build/build-gcc-linux-stage2/gcc/testsuite/g++/../../xg++ -B/var/lib/jenkins/workspace/riscv64-gnu-toolchain/Build/build-gcc-linux-stage2/gcc/testsuite/g++/../../ /var/lib/jenkins/workspace/riscv64-gnu-toolchain/riscv-gnu-toolchain/riscv-gcc/gcc/testsuite/c-c++-common/Warray-bounds-2.c  -march=rv64imafdc -mabi=lp64d -mcmodel=medlow   -fno-diagnostics-show-caret -fno-diagnostics-show-line-numbers -fdiagnostics-color=never  -nostdinc++ -I/var/lib/jenkins/workspace/riscv64-gnu-toolchain/Build/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libstdc++-v3/include/riscv64-unknown-linux-gnu -I/var/lib/jenkins/workspace/riscv64-gnu-toolchain/Build/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libstdc++-v3/include -I/var/lib/jenkins/workspace/riscv64-gnu-toolchain/riscv-gnu-toolchain/riscv-gcc/libstdc++-v3/libsupc++ -I/var/lib/jenkins/workspace/riscv64-gnu-toolchain/riscv-gnu-toolchain/riscv-gcc/libstdc++-v3/include/backward -I/var/lib/jenkins/workspace/riscv64-gnu-toolchain/riscv-gnu-toolchain/riscv-gcc/libstdc++-v3/testsuite/util -fmessage-length=0  -std=gnu++14 -O2 -Warray-bounds -Wno-stringop-overflow  -S   -o Warray-bounds-2.s

This emits Warray-bounds-2.s file.So, we want to know how the decision of PASS/FAIL is determined and also how individual fail can be reproduced?

The logs are attached for reference

Thanks
[g++_dot_sum.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3600640/g%2B%2B_dot_sum.txt)

[g++_dot_log.tar.gz](https://github.com/riscv/riscv-gnu-toolchain/files/3600651/g%2B%2B_dot_log.tar.gz)

jim-wilson, 2019-09-11:
PASS/FAIL depends on the testcase.  Some are compile only.  Some are execute.  Some are checking to see if a particular instruction is or is not emitted.  Some are checking to see if a particular warning is or is not emitted.  Etc.  If a testcase is in a dir named compile or execute then it is a simple compile or execute testcase.  If the testcase is in a dg dir, then there will be comment in the testcase that give instructions to the testsuite driver for actions to perform and conditions to check.  You can also check the *.exp file in the same dir as the testcase to see what actions will be performed.

This particular testcase has dg comments that explains how it works.

You already have the command to reproduce.

I don't have infinite amounts of time available to do infinite amounts of work, so I won't be analyzing this.  You could try analyzing it yourself.

I can say one thing though, I have seen this testcase fail when a buggy native gcc was used to build the riscv cross compiler.  I think it was the Ubuntu 16.04 gcc-5.4 that was causing the trouble.  But I haven't seen these failures recently, so I think Ubuntu might have fixed the x86 gcc bug, or maybe the riscv gcc sources changed enough that they aren't triggering the x86 gcc bug anymore.

TommyMurphyTM1234, 2022-09-29:
No update/activity in over three years.

closed 2022-09-29 by TommyMurphyTM1234