riscv-collab/riscv-gnu-toolchain #507: can I compile python code for RISCV32I ??

Mohsannaeem, 2019-09-14:
I had install riscv-gnu-toolchain for c code and now i wondering could i compile my python code to generate Machine code  for riscv...?

jim-wilson, 2019-09-15:
GCC has no support for compiling python code.  You would need some python compiler instead.

I don't know if you can compile python code to machine code, but you would need some python compiler for that.  And I don't know if cross compilation is a possibility.  I also don't know if any python compiler if it exists, has been ported to RISC-V.

If you can convert python to some other language like C++, then you could use gcc to compile that.  it might be simplest to just boot a linux distro on qemu and then do the python compiling natively.

carlosdelfino, 2021-06-29:
I'm also looking for similar alternatives. The closest I got was with Chocopy.org. But they don't provide an open compiler.

carlosdelfino, 2021-06-29:
Correcting me, a version of Venus tuned for Chocopy is made available at https://github.com/chocopy/venus.