riscv-collab/riscv-gnu-toolchain #508: can't print float

ppcust, 2019-09-17:
I use sifive latest riscv tool chain for generate float print , but result is not right. code as follow

 char printbuffer[256];
void F_PSPRINTF ( const char * fmt, ... )
{
    va_list args;
    int n;
    va_start ( args, fmt );
    n = vsprintf ( printbuffer, fmt, args );
    va_end ( args );
	SerialOutString(printbuffer);
}

  F_PSPRINTF("aa %f\n",1.123);

always 0.000

aswaterman, 2019-09-17:
Is there any chance that you called F_PSPRINTF from a context where the
function wasn't declared--i.e. did you get an "implicit declaration"
warning?  On RISC-V, varargs functions use a different calling convention,
so if the function isn't declared with the correct prototype, the FP arg
won't be passed correctly.

ppcust, 2019-09-17:
thank you for reminding me,  i got an " implicit declaration warning", 
after I remove this warning,  the result is right. 

thanks a lot~~~

aswaterman, 2019-09-17:
👍

closed 2019-09-17 by aswaterman