riscv-collab/riscv-gnu-toolchain #509: support for macOS

sstaub, 2019-09-17:
Try to use GD32 (Sipeed Longan Nano) with PlatformIO on macOS with no luck.
Can you add support?
[image]

jim-wilson, 2019-09-17:
This doesn't look like a riscv-gnu-toolchain problem.

jim-wilson, 2019-09-17:
You can find macos toolchain binaries on the SiFive web site, but I don't know if that will help you, as this isn't a riscv-gnu-toolchain problem.

cghalibkhan, 2019-10-14:
HI guys,

         From SiFive web site we are getting riscv64-unknown-elf toolchain.
* But I need riscv64-unknown-linux-gnu-  toolchain, 
To Build riscv64-unknown-linux-gnu-  toolchain, I Followed this step's:
1. I cloned riscv sources from git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
2. ./configure --prefix=MY_PATH
3. make linux

**Here I was facing 2 issues,**

4. 
                                        1'st issue is :
                                -------------------------
Linux Tool Build binutiles & gcc But glibc is failing to build, Error is related to make version as follows.

*** These critical programs are missing or too old: make
*** Check the INSTALL file for required versions.
make: *** [Makefile:252: stamps/build-glibc-linux-rv64imafdc-lp64d] Error 1

Here I was using latest make of 4.2 but still getting this Error, can anyone help me to resolve this issue.

Here sysroot is not build, Because of glibc faillure, so I copied  ubuntu build riscv sysroot to Mac it was working, But still I need to build glibc on MACos Machine.

                                           2'nd issues is :
                                  -----------------------------
riscv64-unknown-linux-gnu-c++ is not building, To resolve this issue I configured tool with --enable-languages=c,c++ 

But still  riscv64-unknown-linux-gnu-c++ is not building 

Can anyone help me to resolve this issue.

Thanks,
Ghalib khan.

jim-wilson, 2019-10-14:
On Mon, Oct 14, 2019 at 7:56 AM cghalibkhan <notifications@github.com>
wrote:

>
>    1.
>
>                                     1'st issue is :
>                             -------------------------
>
>
> Linux Tool Build binutiles & gcc But glibc is failing to build, Error is
> related to make version as follows.
>
> *** These critical programs are missing or too old: make
>

I don't have a mac, but know that building stuff there can be tricky.  The
shell code glibc uses to check the make version number is
ac_prog_version=`$MAKE --version 2>&1 | sed -n 's/^.*GNU
Make[^0-9]*\([0-9][0-9.]*\).*$/\1/p'`

If make is GNU make, and the version number is correct (4.1 or later), then
maybe the problem is with sed.

In fact, looking at the sifive/freedom-tools which has macos support (but
not linux toolchain support), I see the makefile has for darwin (aka macos)
 LIBTOOLIZE ?= glibtoolize
TAR ?= gtar
SED ?= gsed
AWK ?= gawk

>                                        2'nd issues is :
>                               -----------------------------
>
> riscv64-unknown-linux-gnu-c++ is not building, To resolve this issue I
> configured tool with --enable-languages=c,c++
>
You didn't report the error here.  C++ is configured and built by default
for linux toolchains, so something else is wrong, but the suggestion above
might help.

Jim

ilg-ul, 2019-10-14:
The GCC auto-configure scripts do not support the specifics of Darwin, which is a BSD derivative that has little to do with GNU.

If you want to build GCC yourself on macOS you need to recreate a GNU environment, and probably the easiest way is via Homebrew.

For details how to do it and what packages (tools and libraries) are required to build GCC, Google should be able to provide some good starting points.

cghalibkhan, 2019-10-16:
> On Mon, Oct 14, 2019 at 7:56 AM cghalibkhan ***@***.***> wrote: 1. 1'st issue is : ------------------------- Linux Tool Build binutiles & gcc But glibc is failing to build, Error is related to make version as follows. *** These critical programs are missing or too old: make
> I don't have a mac, but know that building stuff there can be tricky. The shell code glibc uses to check the make version number is ac_prog_version=`$MAKE --version 2>&1 | sed -n 's/^.*GNU Make[^0-9]*\([0-9][0-9.]*\).*$/\1/p'` If make is GNU make, and the version number is correct (4.1 or later), then maybe the problem is with sed. In fact, looking at the sifive/freedom-tools which has macos support (but not linux toolchain support), I see the makefile has for darwin (aka macos) LIBTOOLIZE ?= glibtoolize TAR ?= gtar SED ?= gsed AWK ?= gawk
> 2'nd issues is : ----------------------------- riscv64-unknown-linux-gnu-c++ is not building, To resolve this issue I configured tool with --enable-languages=c,c++
> You didn't report the error here. C++ is configured and built by default for linux toolchains, so something else is wrong, but the suggestion above might help. Jim

Thanks for your reply.

            Regarding 2nd issue riscv-c++, No Error is coming. The tool is building successfully but riscv-c++ & g++ is not building. Here is the log, I think this will provide you some info.

[clang_with_riscv_release_src_build.log](https://github.com/riscv/riscv-gnu-toolchain/files/3732874/clang_with_riscv_release_src_build.log)

    Please have look at below executable in that riscv64-unknown-linux-gnu-c++ & g++ is not there.

riscv64-unknown-linux-gnu-addr2line
riscv64-unknown-linux-gnu-elfedit
riscv64-unknown-linux-gnu-gcc-ranlib
riscv64-unknown-linux-gnu-gdb-add-index
riscv64-unknown-linux-gnu-objcopy
riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-ar
riscv64-unknown-linux-gnu-gcc	
riscv64-unknown-linux-gnu-gcov
riscv64-unknown-linux-gnu-gprof
riscv64-unknown-linux-gnu-objdump
riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-as
riscv64-unknown-linux-gnu-gcc-9.2.0
riscv64-unknown-linux-gnu-gcov-dump
riscv64-unknown-linux-gnu-ld
riscv64-unknown-linux-gnu-ranlib
riscv64-unknown-linux-gnu-strip
riscv64-unknown-linux-gnu-c++filt
riscv64-unknown-linux-gnu-gcc-ar
riscv64-unknown-linux-gnu-gcov-tool
riscv64-unknown-linux-gnu-ld.bfd
riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-cpp
riscv64-unknown-linux-gnu-gcc-nm
riscv64-unknown-linux-gnu-gdb
riscv64-unknown-linux-gnu-nm
riscv64-unknown-linux-gnu-run
  

please help me to overcome this issue.

Thanks, 
Ghalib khan.

cghalibkhan, 2019-10-16:
> On Mon, Oct 14, 2019 at 7:56 AM cghalibkhan ***@***.***> wrote: 1. 1'st issue is : ------------------------- Linux Tool Build binutiles & gcc But glibc is failing to build, Error is related to make [...]

Regarding the 1st issue, I made a few changes in the makefiles to resolve make version issue. make issue resolved but getting a linker error.

/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-environ.c:68: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `malloc':
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:64: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:64: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:73: undefined reference to `__GI___close_nocancel'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:73: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:79: undefined reference to `__lseek'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:56: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `__strerror_r':
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:153: undefined reference to `__GI___fxstat64'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `_itoa_word':
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/../sysdeps/generic/_itoa.h:76: undefined reference to `__GI___fcntl64_nocancel'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `__strerror_r':
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:178: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:178: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:171: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `__GI___assert_perror_fail':
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:224: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `__strsep':
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:272: undefined reference to `rtld_errno'
/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os:/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf/dl-minimal.c:279: more undefined references to `rtld_errno' follow
[... cut ...]
make[3]: *** [Makefile:508: /Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so] Error 1
make[3]: Leaving directory '/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc/elf'
make[2]: *** [Makefile:259: elf/subdir_lib] Error 2
make[2]: Leaving directory '/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/riscv-glibc'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/Users/macagent/khan/clang-r1.0/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d'
make: *** [Makefile:253: stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2

kito-cheng, 2019-10-16:
Hi @cghalibkhan:

Could you check your file system is case-sensitive or case-insensitive? glibc must build on case-sensitive system, those error message seems like I got on case-insensitive file system before.

robertlipe, 2020-12-25:
In case it helps anyone else reading this issue (does anyone read open issues that have been untouched for > a year?) that macro-level problem is known and is being ignored by the Sipeed team responsible for that BSP.  They've pushed back to Nuclei, designer of the core, but it's been frozen for a while.

https://github.com/sipeed/platform-gd32v/issues/6 It's been duped dozens of times, so they've heard the request.

I've found joy using the Homebrew build on GD32V.  Even GDB works great on Segger Mini. You could probably teach Platform IO to call this build by shimming shell scripts to call the Homebrew version or knocking up the path or whatever so the IDE calls the "real" GNU chain provided by Homebrew.

TommyMurphyTM1234, 2022-09-27:
Issues relating to building the toolchain on macOS discussed and clarified. No RISC-V GCC toolchain issue identified.

closed 2022-09-27 by TommyMurphyTM1234