riscv-collab/riscv-gnu-toolchain #516: _gettimeofday__ function

salmansheikh, 2019-09-30:


jim-wilson, 2019-09-30:
gettimeofday requires target and/or OS specific support, and the elf compiler does not supply that, the BSP supplies that.  It does look like you are linking with some Microsemi BSP, so this is a problem that the Microsemi BSP does not support gettimeofday.  It probably does supply other functionality that you can use to implement gettimeofday though, as the benchmark doesn't actually need to know the current time of day, it only needs to be able to get two timer values that it can subtract to compute the execution time.  Also, if you look at the benchmark sources, you will see that it probably has support for more than one timer call, and it may already have support to call a timer function that the Microsemi BSP does support if you define the right macro at compile time.

Worst case, if there is no timer call in the Microsemi BSP you can use, then you can always try reading one of the cycle counter CSRs, and then dividing by the clock frequency.

TommyMurphyTM1234, 2022-09-27:
Not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234