riscv-collab/riscv-gnu-toolchain #518: getting error ----> undefined reference to `getrusage'

billasurendra, 2019-10-01:
I am trying to cross compiling linpackc program with riscv64-unknown-elf-gcc compiler, I am getting undefined reference to `getrusage' error. When I am cross compiling with riscv64-unknown-linux-gnu-gcc i am not getting any error. Can any one please give the technical reason why with elf gcc I am getting error.

Thanks
Billa Surendra

kito-cheng, 2019-10-01:
`getrusage` is POSIX specific function which not provided by newlib.

kito-cheng, 2019-10-01:
Write a dummy `getrusage` if it's not matter for your program, and link your program with that.

ajf58, 2022-01-02:
I think I good example of how to resolve this issue has been provided by @kito-cheng and the issue has been inactive for over two years. Consequently I suggest that this issue is closed.

closed 2022-01-02 by jim-wilson