riscv-collab/riscv-gnu-toolchain #520: la instruction generating incorrect assembly instructions

tdene, 2019-10-03:
When using either one of these commands to compile assembly code:
riscv32-unknown-elf-as -aln -march=rv32i
riscv64-unknown-elf-as -aln -march=rv32i

la instructions do not get converted to an auipc and addi command as they should.

Instead, they get converted to an auipc and mv command.

Here is an example, taken from the riscv-tests jalr test:

la t1, target_2

converts to

auipc ra,0x0
mv ra,ra

instead of

auipc ra,0x0
addi ra,ra,*insert some numerical value here*

aswaterman, 2019-10-03:
The MV pseudoinstructio  is equivalent to ADDI with an immediate operand of
0.

I assume you are disassembling an object file rather than a linked program.
You’ll see that the right thing happens after linking.

tdene, 2019-10-04:
Got it and makes sense, thanks!

closed 2019-10-04 by tdene