riscv-collab/riscv-gnu-toolchain #521: Compile error, unable to use GNU C++ compiler

yujie6, 2019-10-08:
I clone the repo by `git clone --recursive https://github.com/riscv/riscv-gnu-toolchain`
The configuration is
```
./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32
```
And I use `make linux` to build, which produced the following infomation:
```
rm -rf stamps/build-glibc-linux-rv32i-ilp32 build-glibc-linux-rv32i-ilp32
mkdir build-glibc-linux-rv32i-ilp32
cd build-glibc-linux-rv32i-ilp32 && \
	CC="riscv32-unknown-linux-gnu-gcc " \
	CXX="riscv32-unknown-linux-gnu-g++ " \
	CFLAGS=" -mcmodel=medlow -g -O2 " \
	CXXFLAGS=" -mcmodel=medlow -g -O2 " \
	ASFLAGS=" -mcmodel=medlow " \
	/home/yujie6/Downloads/riscv-gnu-toolchain/riscv-glibc/configure \
	--host=riscv32-unknown-linux-gnu \
	--prefix=/usr \
	--disable-werror \
	--enable-shared \
	--enable-obsolete-rpc \
	--with-headers=/home/yujie6/Downloads/riscv-gnu-toolchain/linux-headers/include \
	--disable-multilib \
	--enable-kernel=3.0.0 \
	 \
	--libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv32-unknown-linux-gnu-gcc 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv32-unknown-linux-gnu-gcc  accepts -g... yes
checking for gcc... gcc
checking for riscv32-unknown-linux-gnu-readelf... riscv32-unknown-linux-gnu-readelf
checking whether we are using the GNU C++ compiler... no
checking whether riscv32-unknown-linux-gnu-g++  accepts -g... no
checking whether riscv32-unknown-linux-gnu-g++  can link programs... no
checking for sysdeps preconfigure fragments... aarch64 alpha arm csky hppa i386 m68k microblaze mips nios2 powerpc riscv glibc requires the A extension
make: *** [Makefile:242：stamps/build-glibc-linux-rv32i-ilp32] error 1
```

kito-cheng, 2019-10-08:
Hi @yujie6:

RISC-V Linux require A-extension as minimal requirement, so at least you should use --wtih-arch=rv32ia or --with-arch=rv64ia for building linux toolchain.

yujie6, 2019-10-08:
Thanks a lot @kito-cheng! It works.

closed 2019-10-08 by yujie6

dhruvguptadg, 2020-08-06:
Hello,

I am having similar problem but i have one question if i change my arch to rv32ia won't it affect my design or anything as i am only using the RV32i version of riscv. 

My second question is if i have encountered this error do i need to uninstall or delete something from the installed toolchain? if yes then how as i don't have much space left in my virtualmachine.

Thanks,
Dhruv Gupta

kito-cheng, 2020-08-06:
> I am having similar problem but i have one question if i change my arch to rv32ia won't it affect my design or anything as i am only using the RV32i version of riscv.

I believe linux can't run without `A` extension, so maybe bare-metal toolchain is what you need.

> My second question is if i have encountered this error do i need to uninstall or delete something from the installed toolchain? if yes then how as i don't have much space left in my virtualmachine.

No good solution for this, but you could try to remove the folder path you pass to `--prefix=`.