riscv-collab/riscv-gnu-toolchain #523: glibc build errors with undefined references in files under elf/

snat99-hub, 2019-10-16:
Build fails with Ubuntu Linux on dual-boot PC and with MS Windows 10 Linux Shell.

Successfully builds standard gnu toolchain binaries under /bin but fails with undefined references when building glibc, when compiling files under elf dir as shown below:

make[5]: Leaving directory 'RISCV_DIR/riscv-glibc/time'
mv -f RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/rtld-libc.aT RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/rtld-libc.a
make[4]: Leaving directory 'RISCV_DIR/riscv-glibc/elf'
riscv64-unknown-linux-gnu-gcc    -nostdlib -nostartfiles -r -o RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os '-Wl,-(' RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/dl-allobjs.os RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/rtld-libc.a -lgcc '-Wl,-)' \
	  -Wl,-Map,RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os.map

riscv64-unknown-linux-gnu-gcc    -nostdlib -nostartfiles -shared -o RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so.new		\
	  -Wl,-z,combreloc -Wl,-z,relro -Wl,--hash-style=both -Wl,-z,defs 	\
	  RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os -Wl,--version-script=RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/ld.map		\
	  -Wl,-soname=ld-linux-riscv64-lp64d.so.1			\
	  -Wl,-defsym=_begin=0

RISCV_DIR/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `process_envvars':
RISCV_DIR/riscv-glibc/elf/rtld.c:2533: undefined reference to `__GI___open64_nocancel'

RISCV_DIR/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `dl_main':
RISCV_DIR/riscv-glibc/elf/rtld.c:1195: undefined reference to `__access'
................
................
RISCV_DIR/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: RISCV_DIR/riscv-glibc/elf/dl-minimal.c:69: undefined reference to `__GI___close_nocancel'
................
................

RISCV_DIR/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os: in function `__strsep':
RISCV_DIR/riscv-glibc/elf/dl-minimal.c:265: undefined reference to `rtld_errno'
RISCV_DIR/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/librtld.os:RISCV_DIR/riscv-glibc/elf/dl-minimal.c:272: more undefined references to `rtld_errno' follow
RISCV_DIR/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so.new: hidden symbol `__libc_check_standard_fds' isn't defined
RISCV_DIR/bin/../lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
Makefile:508: recipe for target 'RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so' failed
make[3]: *** [RISCV_DIR/build-glibc-linux-rv64imafdc-lp64d/elf/ld.so] Error 1

jim-wilson, 2019-10-16:
This problem doesn't look familiar.  You didn't mention which branch you are building (master?), or how you configured, or which Ubuntu version you are building on, but I'm skeptical that this will help me reproduce the problem.  You might want to check for unexpected problems like a disk full error causing corrupted files that need to be deleted in order to get a correct build.  That is my best guess at the moment.  If the glibc build was corrupted you can rm -rf build-glibc-* and rm stamps/build-glibc-* and try the build again.

snat99-hub, 2019-10-16:
Environment:

MS provided Linux Shell in Windows 10: Ubuntu 18.04 LTS

I followed the standard build steps in README in this github site:

1) git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
2) ./configure --prefix=/xyz/riscv
3) make MAKEINFO=true linux
(wouldn't build w/o the makeinfo specification even though Texinfo is installed)

After build of /bin:
riscv64-unknown-linux-gnu-gcc --version:
        riscv64-unknown-linux-gnu-gcc (GCC) 9.2.0

Tried the build after the rms as suggested above:
 (rm -rf build-glibc-* and rm stamps/build-glibc-* )
and also after a make clean - got the same results.

I don't have access to any of:
a) corporate X86 system running Linux
b) an X86 machine with Ubuntux Linux pre-installed or
c) a bare-bones X86 with a direct install of Linux.

I have managed to install a working older version of risc-v gnu toolchain under Cygwin:
    riscv64-unknown-elf-gcc.exe (GCC) 7.1.1 20170509
The problem is I could not build Spike under Cygwin, but was able to under Ubuntu Linux.
Most of the other risc-v tools seem to expect a "pure" Linux env in order to build.

Would you able to perform the above riscv-gnu-toolchain build process on a standard Windows 10 laptop, using the Ubuntu shell provided by MS under Windows 10 to see if the problem can be replicated?

Thanks for any help.

snat99-hub, 2019-10-17:
No help received in the last day. Since the rest of the Riscv tool ecosystem depends on the GCC cross-compiler working, this is pretty much a dead stop.

From what I can tell, what the loader complains as undefined vars are declared extern in files under glibc/elf but are actually defined in .c files under glibc/io . So the question is what in the construction of the Makefile is causing them not to be linked together ?

Anybody can help?

jim-wilson, 2019-10-17:
I don't have access to windows in the office, and it is basically just me trying to help people.  Though in another issue Kito Cheng pointed out that a glibc build can fail on MacOS if you are on a case-insensitive filesystem.  I would expect the Windows Linux subsystem to have a case-sensitive filesystem, but apparently this is configurable.  So maybe you are accidentally on a case-insensitive filesystem, and it will work if you change the filesystem configuration to case-sensitive.

aswaterman, 2019-10-18:
Yes, this seems like it could be the same case-insensitivity issue.  The
specific problem is that glibc creates some object files with the .os
extension, and creates other object files with the .oS extension, and some
of these files have the same base name.  If they stomp on each other, the
wrong objects will get linked, causing the kind of failure you're seeing.

snat99-hub, 2019-10-21:
Yes, thanks for catching this, this was the problem.
Windows 10 case-sensitivity is apparently settable on a per file or per directory basis with the following command:

fsutil.exe  file setCaseSensitiveInfo C:\<foldername> enable

However this does not apply recursively, which is useless, since the builds do create directory trees.

Anyway the naming of nonshared files with a .oS, with lots of .os files lying around is absurd - its not you like you are running out of filenamespace. I replaced all the .oS extensions with .ononshared in all the config* Make* and *.mk files found recursively under riscv-glibc, and that works. I suggest this change be made to avoid further bite-backs.

aswaterman, 2019-10-21:
I agree this is an unnecessary complication.  It's a generic glibc thing, though, not a RISC-V thing, so if you want to pursue changing it, you should bring it up on the glibc mailing list.

expnkx, 2020-10-17:
@snat99-hub I try to build an x86_64 Linux cross-toolchain with msys2. I also meet glibc build problems. I have modified .oS to .ononshared but the compile still fails. Do you know why?

ild-glibc/elf/rtld-libc.a -lgcc '-Wl,-)' \
          -Wl,-Map,/build-glibc/elf/librtld.os.map
x86_64-linux-gcc   -nostdlib -nostartfiles -shared -o /build-glibc/elf/ld.so.new                \
          -Wl,-z,combreloc -Wl,-z,relro -Wl,--hash-style=both -Wl,-z,defs       \
          /build-glibc/elf/librtld.os -Wl,--version-script=/build-glibc/ld.map          \
          -Wl,-soname=ld-linux-x86-64.so.2                      \
          -Wl,-defsym=_begin=0
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: in function `dlmopen_doit':
D:\msys64\glibc\elf/rtld.c:655: undefined reference to `__environ'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: in function `process_dl_debug':
D:\msys64\glibc\elf/rtld.c:2542: undefined reference to `memcmp'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\elf/rtld.c:2588: undefined reference to `_exit'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\elf/rtld.c:2553: undefined reference to `strnlen'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\elf/rtld.c:2553: undefined reference to `memcpy'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: in function `version_check_doit':
D:\msys64\glibc\elf/rtld.c:682: undefined reference to `_exit'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: in function `audit_list_next':
D:\msys64\glibc\elf/rtld.c:244: undefined reference to `strcspn'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\elf/rtld.c:247: undefined reference to `memcpy'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: in function `dso_name_valid_for_suid':
D:\msys64\glibc\elf/rtld.c:179: undefined reference to `strlen'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\elf/rtld.c:180: undefined reference to `memchr'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: in function `handle_preload_list':
D:\msys64\glibc\elf/rtld.c:898: undefined reference to `strcspn'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\elf/rtld.c:901: undefined reference to `memcpy'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: in function `dso_name_valid_for_suid':
D:\msys64\glibc\elf/rtld.c:179: undefined reference to `strlen'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\elf/rtld.c:180: undefined reference to `memchr'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: in function `process_envvars':
D:\msys64\glibc\elf/rtld.c:2601: undefined reference to `_environ'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/elf/librtld.os: relocation R_X86_64_PC32 against undefined hidden symbol `_environ' can not be used when making a shared object
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
make[2]: *** [Makefile:567: /build-glibc/elf/ld.so] Error 1
make[2]: Leaving directory '/glibc/elf'
make[1]: *** [Makefile:479: elf/subdir_lib] Error 2
make[1]: Leaving directory '/glibc'
make: *** [Makefile:9: all] Error 2

jim-wilson, 2020-10-17:
There is probably some other linux dependency you missed, or maybe the oS to ononshared change wasn't complete and correct.

I'd suggest using WSL2 to boot linux and build a toolchain inside linux.  You could use this to cross build a toolchain that runs on linux.  Or just build a linux hosted cross toolchain.

expnkx, 2020-10-17:
> There is probably some other linux dependency you missed, or maybe the oS to ononshared change wasn't complete and correct.
>
> I'd suggest using WSL2 to boot linux and build a toolchain inside linux. [...]

I do not want to use wsl2 since I already have multiple linux hosted toolchains. Using a windows based linux toolchains save my time and resources since I can test my programs faster and easier.

WSL2 is also a waste of resources since it boots a virtual machine.

expnkx, 2020-10-17:
> There is probably some other linux dependency you missed

I've built the cross-toolchain with the same procedure under linux and even newlib based mingw-w64 toolchains. The kernel header was copied from another linux environment that generates the header files manually.

> or maybe the oS to ononshared change wasn't complete and correct.

I think another issue with windows is that CreateProcess restricts the maximum characters you can put into the parameters list. That might create issues?

expnkx, 2020-10-17:
Another problem is
make -f /build-glibc/elf/librtld.mk -f rtld-Rules
make[3]: Entering directory '/glibc/elf'
rtld-Rules:40: *** This makefile is a subroutine of elf/Makefile not to be used directly.  Stop.
make[3]: Leaving directory '/glibc/elf'
make[2]: *** [Makefile:554: /build-glibc/elf/rtld-libc.a] Error 2
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/glibc/elf'
make[1]: *** [Makefile:479: elf/subdir_lib] Error 2
make[1]: Leaving directory '/glibc'
make: *** [Makefile:9: all] Error 2

expnkx, 2020-10-17:
it looks this generates empty file.

mv -f /build-glibc/elf/librtld.mapT /build-glibc/elf/librtld.map
LC_ALL=C \
sed -n 's@^[0-9a-f ]*/build-glibc/\([^(]*\)(\([^)]*\.os\)) *.*$@\1 \2@p' \
    /build-glibc/elf/librtld.map | \
while read lib file; do \
  case $lib in \
  libc_pic.a) \
    LC_ALL=C fgrep -l /$file \
          /build-glibc/stamp.os /build-glibc/*/stamp.os | \
    LC_ALL=C \
    sed 's@^/build-glibc/\([^/]*\)/stamp\.os$@rtld-\1'" +=$file@"\
    ;; \
  */*.a) \
    echo rtld-${lib%%/*} += $file ;; \
  *) echo "Wasn't expecting $lib($file)" >&2; exit 1 ;; \
  esac; \
done > /build-glibc/elf/librtld.mkT
echo rtld-subdirs = `LC_ALL=C sed 's/^rtld-\([^ ]*\).*$/\1/' /build-glibc/elf/librtld.mkT \
                     | LC_ALL=C sort -u` >> /build-glibc/elf/librtld.mkT
mv -f /build-glibc/elf/librtld.mkT /build-glibc/elf/librtld.mk
make -f /build-glibc/elf/librtld.mk -f rtld-Rules
make[3]: Entering directory '/glibc/elf'
rtld-Rules:40: *** This makefile is a subroutine of elf/Makefile not to be used directly.  Stop.
make[3]: Leaving directory '/glibc/elf'
make[2]: *** [Makefile:554: /build-glibc/elf/rtld-libc.a] Error 2
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/glibc/elf'
make[1]: *** [Makefile:479: elf/subdir_lib] Error 2
make[1]: Leaving directory '/glibc'
make: *** [Makefile:9: all] Error 2

expnkx, 2020-10-17:
```cpp
unlvs@DESKTOP-DFHPDC1 MINGW64 /build-glibc
$ x86_64-linux-gcc   -shared -static-libgcc -Wl,-O1  -Wl,-z,defs -Wl,-dynamic-linker=/x86_64-linux/x86_64-linux/lib/ld-linux-x86-64.so.2  -B/build-glibc/csu/  -Wl,--version-script=/build-glibc/libmvec.map -Wl,-soname=libmvec.so.1 -Wl,-z,combreloc -Wl,-z,relro -Wl,--hash-style=both   -L/build-glibc -L/build-glibc/math -L/build-glibc/elf -L/build-glibc/dlfcn -L/build-glibc/nss -L/build-glibc/nis -L/build-glibc/rt -L/build-glibc/resolv -L/build-glibc/mathvec -L/build-glibc/support -L/build-glibc/crypt -L/build-glibc/nptl -Wl,-rpath-link=/build-glibc:/build-glibc/math:/build-glibc/elf:/build-glibc/dlfcn:/build-glibc/nss:/build-glibc/nis:/build-glibc/rt:/build-glibc/resolv:/build-glibc/mathvec:/build-glibc/support:/build-glibc/crypt:/build-glibc/nptl -o /build-glibc/mathvec/libmvec.so -T /build-glibc/shlib.lds /build-glibc/csu/abi-note.o -Wl,--whole-archive /build-glibc/mathvec/libmvec_pic.a -Wl,--no-whole-archive /build-glibc/math/libm.so.6   -Wl,--start-group /build-glibc/libc.so /build-glibc/libc_nonshared.a -Wl,--as-needed /build-glibc/elf/ld.so -Wl,--no-as-needed -Wl,--end-group
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_log2_core_sse4.os): in function `_ZGVbN2v_log_sse4':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_log2_core_sse4.S:214: undefined reference to `log'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_log2_core_sse4.S:224: undefined reference to `log'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_log4_core_avx2.os): in function `_ZGVdN4v_log_avx2':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_log4_core_avx2.S:194: undefined reference to `log'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_log4_core_avx2.S:205: undefined reference to `log'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_log8_core_avx512.os): in function `_ZGVeN8v_log_knl':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_log8_core_avx512.S:225: undefined reference to `log'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_log8_core_avx512.os):D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_log8_core_avx512.S:233: more undefined references to `log' follow
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_exp2_core_sse4.os): in function `_ZGVbN2v_exp_sse4':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_exp2_core_sse4.S:210: undefined reference to `exp'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_exp2_core_sse4.S:220: undefined reference to `exp'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_exp4_core_avx2.os): in function `_ZGVdN4v_exp_avx2':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_exp4_core_avx2.S:196: undefined reference to `exp'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_exp4_core_avx2.S:207: undefined reference to `exp'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_exp8_core_avx512.os): in function `_ZGVeN8v_exp_knl':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_exp8_core_avx512.S:226: undefined reference to `exp'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_exp8_core_avx512.os):D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_exp8_core_avx512.S:234: more undefined references to `exp' follow
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_pow2_core_sse4.os): in function `_ZGVbN2vv_pow_sse4':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_pow2_core_sse4.S:416: undefined reference to `pow'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_pow2_core_sse4.S:427: undefined reference to `pow'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_pow4_core_avx2.os): in function `_ZGVdN4vv_pow_avx2':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_pow4_core_avx2.S:370: undefined reference to `pow'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/svml_d_pow4_core_avx2.S:382: undefined reference to `p [... cut ...]
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:/msys64/build-glibc/mathvec/libmvec_pic.a(svml_d_pow2_core-sse2.os): in function `_ZGVbN2vv_pow_sse2':
D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/../svml_d_pow2_core.S:24: undefined reference to `pow'
d:/msys64/x86_64-linux/bin/../lib/gcc/x86_64-linux/11.0.0/../../../../x86_64-linux/bin/ld.exe: D:\msys64\glibc\mathvec/../sysdeps/x86_64/fpu/multiarch/../svml_d_pow2_core.S:24: undefined reference to `pow'
collect2.exe: error: ld returned 1 exit status

```

I have fixed that issue. However there is a new issue lol.

expnkx, 2020-10-18:
[image]
HAHA, I have done the compilation.

expnkx, 2020-10-18:
wsl2 also wastes resources since it boots a virtual machine while i pay nothing for using a cross toolchain instead.

expnkx, 2020-10-18:
i do not want to use wsl2 since i already have toolchains under linux. That defeats the purpose. I want to use cross toolchain since i build my programs on windows and linux simultaenously. A cross toolchain allows me to compile my program under different platforms very fast instead of a context switch to wsl terminal which wastes my time.

expnkx, 2020-10-18:
This also works for aarch64. I have provided a detailed explanation on how to cross-compile with windows
[image]

https://github.com/expnkx/GCC-cross-compilation-scripts/blob/master/windows%20hosted/aarch64-linux-gnu.txt

Nelson1225, 2020-10-19:
https://github.com/riscv/riscv-gnu-toolchain/issues/742
Looks like a similar problem, but still failed when using WSL2...

expnkx, 2020-10-29:
Hello. I have made a guideline on how to build windows msys2 mingw-w64 cross toolchains for x86_64-linux-gnu, aarch64-linux-gnu and riscv64-linux-gnu.

%20hosted

expnkx, 2020-10-29:
[image]

expnkx, 2020-10-29:
> #742
> Looks like a similar problem, but still failed when using WSL2...

wsl2 lol. why wsl2?

TommyMurphyTM1234, 2022-09-27:
Issue is that the Linux toolchain build requires a case sensitive filesystem which was not originally used.

closed 2022-09-27 by TommyMurphyTM1234