riscv-collab/riscv-gnu-toolchain #524: make report-linux fails

dylanmc, 2019-10-16:
After a good 3 hours, with many passed and surprisingly many "unexpected" failures, `make report-linux` failed here:

```
date > stamps/check-gcc-linux
/home/vagrant/riscv-gnu-toolchain/scripts/testsuite-filter gcc glibc /home/vagrant/riscv-gnu-toolchain/test/whitelist `find build-gcc-linux-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`
Traceback (most recent call last):
  File "/home/vagrant/riscv-gnu-toolchain/scripts/testsuite-filter", line 297, in <module>
    main()
  File "/home/vagrant/riscv-gnu-toolchain/scripts/testsuite-filter", line 288, in main
    unexpected_results)
  File "/home/vagrant/riscv-gnu-toolchain/scripts/testsuite-filter", line 181, in filter_result
    for testtool, variation_unexpected_result in unexpected_results.iteritems():
AttributeError: 'dict' object has no attribute 'iteritems'
Makefile:787: recipe for target 'report-gcc-linux' failed
make: *** [report-gcc-linux] Error 1
```

dylanmc, 2019-10-16:
Just for info, here are the summaries of the tests that did complete:
```
		=== gcc Summary ===

# of expected passes		104371
# of unexpected failures	10
# of unexpected successes	3
# of expected failures		402
# of unsupported tests		1949

		=== g++ Summary ===

# of expected passes		122196
# of unexpected failures	24
# of expected failures		503
# of unsupported tests		6054

		=== gfortran Summary ===

# of expected passes		49352
# of unexpected failures	11
# of expected failures		178
# of unsupported tests		90
```

jim-wilson, 2019-10-16:
The unexpected failures are normal.  There are so many tests, and so many targets, and so many patches being added to gcc, and so few people trying to fix problems, that it is impossible to keep them all working for all targets all of the time.  Most of the RISC-V failures are trivial problems.  For instance there are some tests trying to count nops emitted by a specific feature, but it is confused by the ".option nopic" emitted by the RISC-V gcc and gets the count wrong.  This needs a testsuite fix, but is so unimportant we haven't bothered to try to fix the testsuite yet.  We have a white list to ignore the known failures so that only new failures cause make report to fail, and then we look at new failures and decide if they are important enough to fix, or not important enough that we can add them to the white list.

The python error is not normal.  I think this is a python2 versus python3 issue.  We unfortunately use Ubuntu 16.04 here which uses python2 by default.  Maybe you are using python3 by default?  If you change .iteritems() to .items() does that work?  Running "make report" again won't run the testsuites again, it will only run the code that compares against the white lists.

dylanmc, 2019-10-16:
Yes, it's a python issue - thanks, and glad to see that my tests did finish other than that.  I guess she fix, then, is to change the README.md to clearly specify Python 2. As it is, the only reason I had Python in the VM before this point was to run QEMU, which says it can handle Python 2.7 or 3.

dylanmc, 2019-10-16:
Oh, would it be helpful to share a Vagrantfile I made that runs the build? I can't figure out where the beginner community, and folks who want to help them, hang out wrt. RISC-V. I got downvoted on StackExchange for asking a related question. Sigh.
[Vagrantfile.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3732360/Vagrantfile.txt)

jim-wilson, 2019-10-16:
I'm not familiar with vagrant.  We do have a .travis.yml file in riscv-gnu-toolchain for automated testing via travis, but that fails so often due to resource limitations that it isn't very useful.  We probably aren't paying enough for proper travis support.

I'm not aware of a specific beginner community.  Some people ask questions here and related github repos.  Some people ask questions on the riscv.org mailing lists like sw-dev.  There is a freenode IRC channel #riscv though if someone gets chatty there people tune out.  Some people ask on the FSF mailing lists.  I haven't been following stack overflow.  I can't read everything.

AntMicro wrote a nice beginner guide for RISC-V, though I'm not sure what happened to it.  I found a copy at https://risc-v-getting-started-guide.readthedocs.io/en/latest/ but I had to use a web search to find it.  I don't think we have any convenient links to it.

dylanmc, 2019-10-16:
Jim,

Thanks for the pointers!  I understand you're part of a small company, and
I appreciate your time.  I think the RISC-V foundation would find that
helping onboard newcomers like myself would pay off nonlinearly.  I'll do
what I can, but I don't even know where I'd leave breadcrumbs to help
people follow my footsteps.

Vagrant is super-easy for end-users to get going, and my vagrantfile
attached to the ticket guides a freshly installed machine all the way
through building the tools and running the tests.  Beginners aren't likely
to have travis infrastructure, even though that's critical for a company to
coordinate all the configurations you need to keep building cleanly.

cheers & thanks!
dylan

kito-cheng, 2019-10-16:
@dylanmc Thanks for report that, I'll fix the compatibility issue between python 2 and 3 :)

jim-wilson, 2019-10-16:
There is no ssh command in riscv-gnu-toolchain.  I think that is a vagrant implementation issue.  But the fact that make report failed means we must have a problem with our white lists being out of date.

RISC-V Foundation doesn't have any engineering staff, or money to hire engineers, so everything is effectively done by volunteers from foundation members, and there aren't enough volunteers for everything that needs to be done.

dylanmc, 2019-10-16:
Understood.

jim-wilson, 2019-10-16:
I tried a multilib linux report on Ubuntu 16.04.  I see 32 c-c++-common/Warray-bounds-2.c failures, but these are known to be due to a bug in the Ubuntu gcc-5.4 that comes with Ubuntu 16.04, this is why the  .travis.yml file specifies gcc6.  I also see one g++.dg/tls/thread_local4g.C.  This one works about 90% of the time when I run it by hand, so this is a user-mode qemu thread support issue not a toolchain issue.  Not everything works in user-mode qemu.  So the Ubuntu 16.04 results look OK, factoring out problems that aren't RISC-V compiler bugs.  These kinds of issues are another reason why getting 0 failures is hard.  I will have to try an Ubuntu 18.04 report run.  There might be other issues there.

dylanmc, 2019-10-17:
The above report was produced with Ubuntu 16.04, since that's what you mentioned is the main platform for your team.

jim-wilson, 2019-10-17:
> The above report was produced with Ubuntu 16.04, since that's what you mentioned is the main platform for your team.

OK.  Did I answer the question then?  You mentioned that make report failed, but didn't mention exactly which tests failed.  It is possible that you might see different failures than me if your environment is different, or if your luck is different and you hit different qemu bugs, etc.

dylanmc, 2019-10-17:
> OK. Did I answer the question then? You mentioned that make report failed, but didn't mention exactly which tests failed. 

I can't tell - but I provided all of the output the make report printed.

> It is possible that you might see different failures than me if your environment is different, or if your luck is different and you hit different qemu bugs, etc.

That could be - I did fresh installs of all of the listed dependencies, but on Ubuntu 16.04, so some of those components could be behaving differently than you're seeing. I think I'm seeing close enough to what you expect that I'll close this one. Thanks for the help!

closed 2019-10-17 by dylanmc

jim-wilson, 2019-10-17:
There should be a bunch of FAIL messages, and then the table that tells you how many fails there were.  You only showed the table not the actual FAILS.