riscv-collab/riscv-gnu-toolchain #530: ilp32f abi not working with -mno-fdiv option

vb000, 2019-10-20:
I was trying to compile a floating point division program with `mno-fdiv` option, using the toolchain compiled with `--with-arch=rv32imaf --with-abi=ilp32f`. I see following error:

```
ld: main.o: in function `.L3':
main.c:(.text.startup+0xb4): undefined reference to `__divsf3'
```

Digging through makefiles, I think the line below is not considering the affect of `-mno-fdiv` option:
https://github.com/riscv/riscv-gcc/blob/bdf3ad8996cb305a822feb1eb11235e08fe4b974/libgcc/config/riscv/t-softfp32#L21

jim-wilson, 2019-10-20:
This feature probably hasn't worked in years.  Does anyone have hardware with FP but not FP div?

Anyways, there is probably a way to fix it by changing the RISC-V softfp config, though we probably don't want to build all of the functions when hard float support is present, just the divide one.

vb000, 2019-10-20:
Our architecture doesn't support FP div and FP sqrt instructions because of area reasons...

If we have hardware FP div and not FP sqrt, for instance, we'd still be forced to use `-mno-fdiv` right? Which would cause compiler to generate soft-float emulations instead of FP instructions. Wouldn't this limitation still be an issue?

jim-wilson, 2019-10-20:
Yes, if you have div but not sqrt there is still a problem, though not as big of a problem.  It is easy to add a -mno-fsqrt option, and sqrt() is in the C library so it is there regardless of whether you have a sqrt instruction.  There is a potential problem if the C library sqrt() function uses the sqrt instruction, but that would have to be handled in the C library, as we don't have a libgcc routine for sqrt.

aswaterman, 2019-10-20:
It’s tangential to this issue, but consider that a one-bit-at-a-time
single-precision fdiv+fsqrt unit can be built in a very small footprint (<
1K gates if the normalization shifters are shared with other FP units). So
in the long term, it’s probably better to provide a slow hardware
implementation.

taylor-bsg, 2019-10-21:
@aswaterman That is an interesting point — would you have a reference or sample code lying around that I can point a student at?

aswaterman, 2019-10-21:
Some cores I've worked on (including Rocket) have used https://github.com/ucb-bar/berkeley-hardfloat for this purpose; there's a unit called DivSqrtRecFN_small.

jim-wilson, 2019-10-24:
I've got a patch that adds the missing divide routines to libgcc, but it is a partial solution, because the libraries (C, math, libstdc++, etc) may still have divide instructions in them.  In order to fix that, we would have to multilib on -mno-fdiv, and it does not seem worthwhile to do that for such an uncommonly used option.  Or alternatively we would have to modify the compiler to make -mno-fdiv the default, so that the libraries are compiled with it.  We could do that with a configure option, or by adding a new architecture extension like Xnofdiv.  The configure option choice is easier, but I haven't tried to do that yet.  We could add a --without-fdiv configure option for instance that works similarly to how --with-cmodel works.

[libgcc-mno-fdiv.txt](https://github.com/riscv/riscv-gnu-toolchain/files/3769564/libgcc-mno-fdiv.txt)

vb000, 2019-10-24:
Thanks for patch @jim-wilson. This is efficient than the patch we did to blanket enable sf routines. Doing that was acceptable as it only increase the library size and not the executable size.

About the second issue you mentioned though, actually, I had found a hack to disable fdiv instructions in libraries. We can override the make variable called `CFLAGS_FOR_TARGET_EXTRA` used here during the toolchain build:
https://github.com/riscv/riscv-gnu-toolchain/blob/3ac0d0544ff2b56765523330a2ff6f19ebc542fa/Makefile.in#L63
We built the toolchain with `make CFLAGS_FOR_TARGET_EXTRA='-mno-fdiv' newlib`. In fact we passed all the flags we wanted to the toolchain build this way. So, having configure options called  `--with-target-cflags`, `--with-target-cppflags` would also be helpful I believe?

jim-wilson, 2019-10-25:
Yes, a configure option to set CFLAGS, I was thinking something -mno-fdiv specific, but we could do something more generic like you suggested.  Using CFLAGS_FOR_TARGET_EXTRA also works, and doesn't need a new configure option.  But a configure option is probably easier to use than remembering to set CFLAGS_FOR_TARGET_EXTRA.

jim-wilson, 2019-11-04:
Should be fixed now, if you use the configure options to add -mno-fdiv to the library builds.

TommyMurphyTM1234, 2022-09-29:
No update/activity in almost three years and @jim-wilson's previous comment states that the original issue was fixed.

closed 2022-09-29 by TommyMurphyTM1234